android-rpi/device_arpi_rpi4 #3: Can't able to boot

skand24, 2020-04-19:
Hello ARPI Team,

I have pulled source code from arpi10 yesterday , followed the same steps written in https://github.com/android-rpi/device_arpi_rpi4 , but can't able to boot .

RaspberryPiFan, 2020-04-19:
What exactly happens when you try to boot?

skand24, 2020-04-20:
I have RPI4 2GB Ram , Its compiled successfully but I have tried to flash , (the flash method is written in above link to my 16GB of SDCARD ), but after when i put into for boot its not booting up .
I have tried to seen watch lsusb but there also no signal came.
Apart from this any other flashing method is there.

RaspberryPiFan, 2020-04-20:
There's no other flashing method, and another person in my discordserver says it works, so I think maybe you did something wrong while flashing. Maybe it helps to re-download everything, and it was just a download error? Also the partitions have to be in the right order, maybe you had the wrong order?

skand24, 2020-04-20:
Hi ,

Can you please check the attached sh file (mkimage.txt) .
I am following the same steps or some more is required .

[mkimg.txt](https://github.com/android-rpi/device_arpi_rpi4/files/4505367/mkimg.txt)

RaspberryPiFan, 2020-04-20:
It seems right to me. Can you please tell me the content of your boot parition and of config.txt and cmdlin.txt on this partition? Just to make sure everything was copied right

skand24, 2020-04-21:
Ok .
Please check the attached files .
Boot Partition Contains : zImage, fixup4.dat, Makefile, star4.elf & all dts /dtsi file

Also I have attached the screenshot of raspeeripi with cmdline & config.txt

[image]

[config.txt](https://github.com/android-rpi/device_arpi_rpi4/files/4510208/config.txt)
[cmdline.txt](https://github.com/android-rpi/device_arpi_rpi4/files/4510209/cmdline.txt)

RaspberryPiFan, 2020-04-21:
Everything seems right to me. Please try to directly flash boot, system.img, vendor.img and userdata onto the sd instead of creating an img file

skand24, 2020-04-22:
Hi Team ,
First Thank you , I can able to boot & able to see the Launcher . Only two apps are there System Tracing & Setting.
I seen few problems and need your guidance.

Initial boot was very slow
System is very slow and when click on particular menu it takes almost 5 second to open.

I'm using 2 GB/16 GB

Not sure only I'm facing this issue or you guys also?

After some time it go to android animation screen(boot screen). And it stuck there.

skand24, 2020-04-22:
And also , I cant able to use adb its not working.

skand24, 2020-04-22:
Does it create any problem if we allocate memory 1GB for system, vendor, boot?
Because the slowness and continue reboot we're seeing.

RaspberryPiFan, 2020-04-23:
That doesn't cause any problem. The slowness is because of the use of an sd card. Do you any external USB drive? If yes, try to install every partition, except boot, on your sd card in a build with this as fstab.rpi4:
```
#<src>                  <mnt_point>         <type>    <mnt_flags and options>                               <fs_mgr_flags>
/dev/block/sda2    /system             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/sda3    /vendor             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/sda4    /data               ext4      noatime,nosuid,nodev,nomblk_io_submit,errors=panic    wait
```
I'm not sure if this works, but you could give it a try. If you want a webbrowser, add `PRODUCT_PACKAGES += Browser2` to rpi4.mk other possible packages are (do the same, but change `Browser2` to the name of the app are: `Email`, `Gallery2`

skand24, 2020-04-24:
Hi I tried multiple times using USB drive but it booted but not loaded the android.
And is this a user build because I cant able to use ADB .
Kindly help me.

RaspberryPiFan, 2020-04-24:
Hav you applied the changes above? (fstab.rpi4)

skand24, 2020-04-24:
Yes and added system.img/vendor img on /dev/sdb1 /dev/sdb2 . But still it didnt worked

RaspberryPiFan, 2020-04-24:
My fault, sorry, try to rebuild with
```
#<src>                  <mnt_point>         <type>    <mnt_flags and options>                               <fs_mgr_flags>
/dev/block/sda1    /system             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/sda2    /vendor             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/sda3    /data               ext4      noatime,nosuid,nodev,nomblk_io_submit,errors=panic    wait
```
or create an empty partition 1 on the usb, system.img on 2 and vendor.img on3, userdata on 4

skand24, 2020-04-26:
Hi,
`/dev/mmcblkp0 /boot 128M` and remaining is unallocated
/dev/sdb1 is empty/unallocated
`/dev/block/sdb2    /system             ext4      ro,barrier=1                                          wait,first_stage_mount`
`/dev/block/sdb3    /vendor             ext4      ro,barrier=1                                          wait,first_stage_mount`
`/dev/block/sdb4    /data               ext4      noatime,nosuid,nodev,nomblk_io_submit,errors=panic    wait`

Can you please tell me this is right configuration. I have tried multiple times but the same issue again again (its booting but android logo is not coming and before it only it restart again).

Please, there is any other chnages I have to do adb because that is also not working.

Kindly help me .

RaspberryPiFan, 2020-04-28:
Then try to go back to sd boot, but try to rebuild it with the latest changes, mybe they were missing in your previus build, after that, try to run `adb shell` and then 
```
dd if=/dev/zero of=/system/swapfile0 bs=1024 count=65536
mkswap /system/swapfile0
swapon /system/swapfile0
```
To add a swap file, so you have more RAM available as swapfile. Please tell me if your system still stays slow after these changes

skand24, 2020-04-29:
For adb shell , adb access should be there .
But adb devices is not showing any device

RaspberryPiFan, 2020-05-01:
Please try to rebuild with `PRODUCT_PACKAGES += Terminal` in rpi4.mk. Now you should have a Terminal installed on the pi where you can run these commands

souvikdatta, 2020-05-17:
Hello, 
I have successfully built  device_arpi_rpi4 from source code by following instructions as mentioned in - https://github.com/android-rpi/device_arpi_rpi4. I am using RPI4 version which has 4 GB of RAM. 

I have added 4 applications in rpi4.mk - 
PRODUCT_PACKAGES += \
    Browser2 \
    Calculator \
    Email \
    Gallery2
With this, when I boot the target, I can see these 4 applications appearing as 4 tiles . However, on clinking on one of these tiles, I see the application opens but then after some time the system resets and I see "android" boot animation. 
Also, I am not sure - once I open an app, how do I exit from that screen. 
Please provide some guidance.

saurabhin25, 2020-05-17:
Can u send me the image?

souvikdatta, 2020-05-17:
Sorry, which Image are you asking?

saurabhin25, 2020-05-17:
Make image (.iso) of ur card

souvikdatta, 2020-05-17:
Well I have not created any iso image - I followed the instruction as mentioned below :-1: # Prepare sd card
 Partitions of the card should be set-up like followings.
  p1 128MB for boot : Do fdisk, set W95 FAT32(LBA) & Bootable type, mkfs.vfat
  p2 768MB for /system : Do fdisk, new primary partition
  p3 128MB for /vendor : Do fdisk, new primary partition
  p4 remainings for /data : Do fdisk, mkfs.ext4
 Set volume label of /data partition as userdata
  : use -L option for mkfs.ext4, and -n option for mkfs.vfat
 
# Write system & vendor partition
  $ cd out/target/product/rpi4
  $ sudo dd if=system.img of=/dev/<p2> bs=1M
  $ sudo dd if=vendor.img of=/dev/<p3> bs=1M
  
# Copy kernel & ramdisk to BOOT partition
  device/arpi/rpi4/boot/* to p1:/
  kernel/arpi/arch/arm/boot/zImage to p1:/
  kernel/arpi/arch/arm/boot/dts/bcm2711-rpi-4-b.dtb to p1:/
  kernel/arpi/arch/arm/boot/dts/overlays/vc4-kms-v3d-pi4.dtbo to p1:/overlays/vc4-kms-v3d-pi4.dtbo
  out/target/product/rpi4/ramdisk.img to p1:/

saurabhin25, 2020-05-17:
> mentioned below 👎 # Prepare sd card
> $ sudo dd if=system.img of=/dev/ bs=1M
> $ sudo dd if=vendor.img of=/dev/ bs=1M

Thats what i am saying just create a image of ur sd card and send it to me

souvikdatta, 2020-05-17:
Well, my card is 32 GB in size and that would take ages for me to share. Are you able to boot your built images successfully on RPI4?

RaspberryPiFan, 2020-05-18:
The boot animation seems broken, try to add 
```
PRODUCT_PROPERTY_OVERRIDES += debug.sf.nobootanimation=1
```
to rpi4.mk to remove it completely, I also experienced the same issue. 
I'll soon provide a prebuilt with some minor improvements, there is still one major bug, but as soon as that bug is fixed (it probably isn't in this repo), the rebuilt image will be available. I think I fixed the navbar, but I'm not sure yet. I'll probably will be able to tell you more tomorrow.

hoangthanh49, 2020-05-25:
hi souvikdatta

I have seen the same problem. After boot, system faced so many crash error. That's why system reset many time.
To know the cause, you have to adb connect and see realtime log.
First, tried to set static IP address for rpi, same IP range with your computer.
Then using below cmd:
adb connect your_IP_of_rpi4:5555

andersling, 2020-08-16:
Hi,all!  I build from the source followed by the instruction of the repository, and the Room and boot upnormally, BUT, THE WIFI IS DISABLED! when I turn the WIFI ON by the setting UI, the OS reboot after a few second, the log is:
--------------
2020-08-12 16:59:23.660 341-358/system_process I/WifiService: setWifiEnabled package=com.android.tv.settings uid=1000 enable=true
2020-08-12 16:59:23.662 341-420/system_process D/WifiController: StaEnabledState.enter()
2020-08-12 16:59:23.662 341-421/system_process D/WifiActiveModeWarden: received a message in WifiDisabledState: { when=0 what=0 target=com.android.internal.util.StateMachine$SmHandler }
2020-08-12 16:59:23.662 341-341/system_process V/SettingsProvider: Notifying for 0: content://settings/global/wifi_on
2020-08-12 16:59:23.662 341-421/system_process D/WifiActiveModeWarden: Switching from WifiDisabledState to ClientMode
2020-08-12 16:59:23.662 341-421/system_process D/WifiActiveModeWarden: Entering ClientModeActiveState
2020-08-12 16:59:23.670 341-421/system_process D/WifiClientModeManager: entering IdleState
2020-08-12 16:59:23.670 341-421/system_process D/WifiActiveModeWarden: State changed from client mode. state = 2
2020-08-12 16:59:23.696 213-213/? I/android.hardware.wifi@1.0-service: Wifi HAL started
2020-08-12 16:59:23.699 341-421/system_process I/WifiVendorHal: Vendor Hal started successfully
2020-08-12 16:59:23.703 341-421/system_process I/android_os_HwBinder: HwBinder: Starting thread pool for getting: android.hidl.manager@1.0::IServiceManager/default
2020-08-12 16:59:23.706 341-421/system_process I/SupplicantStaIfaceHal: Starting supplicant using HIDL
2020-08-12 16:59:23.721 341-421/system_process D/SupplicantStaIfaceHal: Successfully triggered start of supplicant using HIDL
2020-08-12 16:59:23.985 1408-1408/? I/wpa_supplicant: Processing hidl events on FD 3
2020-08-12 16:59:23.992 1408-1408/? I/ServiceManagement: Registered android.hardware.wifi.supplicant@1.2::ISupplicant/default (start delay of 253ms)
2020-08-12 16:59:23.994 1408-1408/? I/wpa_supplicant: Successfully initialized wpa_supplicant
2020-08-12 16:59:24.032 1408-1408/? I/wpa_supplicant: rfkill: Cannot open RFKILL control device
2020-08-12 16:59:24.066 341-360/system_process I/EthernetTracker: interfaceLinkStateChanged, iface: wlan0, up: false
2020-08-12 16:59:24.075 341-360/system_process I/chatty: uid=1000(system) android.fg identical 1 line
2020-08-12 16:59:24.075 341-360/system_process I/EthernetTracker: interfaceLinkStateChanged, iface: wlan0, up: false
2020-08-12 16:59:24.220 341-683/system_process I/android_os_HwBinder: HwBinder: Starting thread pool for getting: android.hardware.wifi.supplicant@1.0::ISupplicant/default
2020-08-12 16:59:24.222 341-683/system_process I/SupplicantStaIfaceHal: Completed initialization of ISupplicant.
2020-08-12 16:59:24.235 213-213/? I/WifiHAL: Initializing wifi
2020-08-12 16:59:24.235 213-213/? I/WifiHAL: Creating socket
2020-08-12 16:59:24.237 213-213/? I/WifiHAL: Initialized Wifi HAL Successfully; vendor cmd = 103
2020-08-12 16:59:24.238 213-213/? I/android.hardware.wifi@1.0-service: Adding interface handle for wlan0
2020-08-12 16:59:24.238 213-213/? W/android.hardware.wifi@1.0-service: No active wlan interfaces in use! Using default
2020-08-12 16:59:24.238 213-213/? E/android.hardware.wifi@1.0-service: Failed to register radio mode change callback
2020-08-12 16:59:24.238 213-213/? W/android.hardware.wifi@1.0-service: No active wlan interfaces in use! Using default
2020-08-12 16:59:24.238 213-213/? E/WifiHAL: Failed to register debug response; result = -95
2020-08-12 16:59:24.238 213-213/? E/android.hardware.wifi@1.0-service: Failed to get driver version: SUCCESS
2020-08-12 16:59:24.238 213-213/? I/android.hardware.wifi@1.0-service: Configured chip in mode 0
2020-08-12 16:59:24.238 213-213/? W/android.hardware.wifi@1.0-service: No active wlan interfaces in use! Using default
2020-08-12 16:59:24.239 341-421/system_process D/HalDevMgr: updateRttController: no one is interested in RTT controllers
2020-08-12 16:59:24.240 213-213/? E/android.hardware.wifi@1.0-service: Failed to set DFS flag; DFS channels may be unavailable.
2020-08-12 16:59:24.252 341-421/system_process D/WificondControl: Setting up interface for client mode
2020-08-12 16:59:24.255 242-242/? I/wificond: create scanner for interface with index: 8
2020-08-12 16:59:24.255 242-242/? I/wificond: subscribe scan result for interface with index: 8
2020-08-12 16:59:24.257 154-154/? I/hwservicemanager: getTransport: Cannot find entry android.hardware.wifi.offload@1.0::IOffload/default in either framework or device manifest.
2020-08-12 16:59:24.257 242-242/? E/wificond: No Offload Service available
2020-08-12 16:59:24.257 242-242/? I/wificond: Offload HAL not supported 
2020-08-12 16:59:24.259 341-421/system_process E/SupplicantStaIfaceHal: Can't call setupIface, ISupplicantStaIface is null
2020-08-12 16:59:24.263 341-421/system_process I/android_os_HwBinder: HwBinder: Starting thread pool for getting: android.hardware.wifi.supplicant@1.0::ISupplicant/default
2020-08-12 16:59:24.275 199-199/? I/netd: interfaceGetCfg("wlan0") <0.33ms>
2020-08-12 16:59:24.276 341-421/system_process I/WifiNative: Interface state changed on Iface:{Name=wlan0,Id=0,Type=STA_CONNECTIVITY}, isUp=true
2020-08-12 16:59:24.277 199-199/? I/netd: interfaceClearAddrs("wlan0") <0.94ms>
2020-08-12 16:59:24.278 199-199/? I/netd: interfaceSetIPv6PrivacyExtensions("wlan0", "true") <0.23ms>
[... cut ...]
2020-08-12 16:59:26.901 199-199/? I/netd: interfaceGetCfg("wlan0") <0.41ms>
2020-08-12 16:59:26.913 1408-1408/? I/wpa_supplicant: wlan0: GAS-QUERY-START addr=48:0e:ec:f0:fd:b6 dialog_token=0 freq=2437
2020-08-12 16:59:26.914 341-421/system_process D/HS20: ANQP initiated on 48:0e:xc:f1:fd:b6
2020-08-12 16:59:26.914 341-421/system_process D/PasspointManager: ANQP entry not found for: 48:0e:ec:f0:fd:b6:<XWIFI_Ling_5G>
-----------------------------------------------------------------
Any idea? thanks a lot!!!

closed 2020-09-11 by peyo-hd