android-rpi/device_arpi_rpi4 #4: WIFI/BT doesn't work

wildpig1979, 2020-05-06:
Hello ARPI Team,
      I followed the steps on README under /device/arpi/rpi4 folder on Android 10 and it could run normally on RPI 4 .  But WIFI/BT doesn't work. I could see wpa_supplicant service was not started. Besides, /dev/socket/wpa_wlan0  and /dev/rfkill was not actuallly created. I could see WIFI firmware under /root/lib/firmware/brcm .  Can you give some suggestions to me ? Is there anything wrong ?

peyo-hd, 2020-05-06:
Wifi/BT is not supported yet.
Waiting for contribution on it.

wildpig1979, 2020-05-07:
Hi Peyo,
     Thank you  for your timely reply.  Looking forward to your contribution.

closed 2020-05-08 by peyo-hd

saurabhin25, 2020-05-08:
Is there any downloadble image

RaspberryPiFan, 2020-05-08:
No, there is **no** downloadable image!!!