android-rpi/device_arpi_rpi4 #6: Image Unstable - Android restarts automatically

souvikdatta, 2020-05-17:
Hello,
I have successfully built device_arpi_rpi4 from source code by following instructions as mentioned in - https://github.com/android-rpi/device_arpi_rpi4. I am using RPI4 version which has 4 GB of RAM.

I have added 4 applications in rpi4.mk -
PRODUCT_PACKAGES +=
Browser2
Calculator
Email
Gallery2
With this, when I boot the target, I can see these 4 applications appearing as 4 tiles . However, on clinking on one of these tiles, I see the application opens but then after some time the system resets and I see "android" boot animation.
Also, I am not sure - once I open an app, how do I exit from that screen.
Please provide some guidance.

peyo-hd, 2020-05-17:
With PC keyboard, F1 is mapped to Home & F2 is mapped to Back.

souvikdatta, 2020-05-17:
Thanks. I just noticed that if I donot connect any display ( I am using a Liliput HDMI display), the system is stable and does not reboot. Once I connect a mouse and display and start interacting with the display, the system reboots. Can you please give a hint, why this is happening?

peyo-hd, 2020-05-18:
It might be unstable since it's for developers to start with.
Wait for bugfix from developers.

closed 2020-05-18 by peyo-hd

RaspberryPiFan, 2020-05-20:
It actually isn't a reset, it just shows the boot animation again. It will disappear after some time.