android-rpi/device_arpi_rpi4 #9: Android Pie on Raspberry Pi4

yagunao999, 2020-05-19:
I want to run Android Pie on Raspberry Pi4.

I downloaded it with the following local manifest with the following contents.
I built it following the instructions in "device_arpi_rpi4".

<? xml version = "1.0" encoding = "UTF-8"?>
<manifest>
 <remote name = "arpi" fetch = "https://github.com/android-rpi" />

<remove-project name = "platform / external / mesa3d" />
<project path = "external / mesa3d" name = "external_mesa3d" revision = "v3d-20.1" remote = "arpi" />
<remove-project name = "platform / external / drm_gralloc" />
<project path = "external / drm_gralloc" name = "external_drm_gralloc" revision = "v3d-20.1" remote = "arpi" />
<remove-project name = "platform / external / libdrm" />
<project path = "external / libdrm" name = "platform / external / libdrm" revision = "pie-cuttlefish-testing" />

<project path = "kernel / arpi" name = "kernel_arpi" revision = "arpi-5.4.y" remote = "arpi" />
<project path = "hardware / arpi" name = "hardware_arpi" revision = "arpi-10" remote = "arpi" />
<project path = "device / arpi / rpi4" name = "device_arpi_rpi4" revision = "arpi-10" remote = "arpi" />

 <project path = "vendor / arpi / RpLauncher" name = "RpLauncher" revision = "pie" remote = "arpi" />

 <remove-project name = "device / google / marlin-kernel" />
 <remove-project name = "device / google / wahoo-kernel" />
 <remove-project name = "device / linaro / hikey-kernel" />
</ manifest>

But it just alternates between the rainbow screen and the black screen.
What should i do?

peyo-hd, 2020-05-19:
device_arpi_rpi4 is targeted to android-10.
No plan to work on android pie. 
Backport for pie could be tried your own forked version.

closed 2020-05-19 by peyo-hd

yagunao999, 2020-05-19:
### 
Thank you.

Is hardware_arpi also targeting android-10?