android-rpi/device_arpi_rpi4 #13: boot fail as system image increases

jaeminkang2, 2020-05-28:
Hi rpi team

I checked the build success & normal boot with this source.
Thank you for the rpi team's effort.

I added another app for testing, and the system area increased by more than 300M.  (add my developing app and  relate google apps)
So, The build was successful by increasing the capacity at BoardConfig.mk, but the boot failed. (Android logo only is visible)

Here's the question.
- If the system image size has increased, we are modifying BoardConfig.mk and modifying the system part size of the sdcard together.
Is there anything else I need to check?

peyo-hd, 2020-05-28:
For partition size change,  modifications you mentioned - BoardConfig.mk & SDcard partition size - are all needed.

jaeminkang2, 2020-05-28:
Thank you for your reply. 
I'm checking a little more.
However, after check it, I'm not going to change the capacity as possible.

closed 2020-05-29 by peyo-hd