android-rpi/device_arpi_rpi4 #17: I can't find video0 under /dev

Midysen, 2020-06-01:
I have successfully built device_arpi_rpi4 from source code by following instructions as mentioned in - https://github.com/android-rpi/device_arpi_rpi4. I am using RPI4 version which has 4 GB of RAM. But,when I use USB camera, I can't find video0 under /dev. Anybody can tell me why?

jaeminkang2, 2020-06-01:
Hi 
That's my stupid question, ^^
When you use the usb camera, did you check the file creation?
I think we need to check the file creation in log first.

Midysen, 2020-06-02:
```
[  485.514957] usb 1-1.1: New USB device found, idVendor=1bcf, idProduct=2283, bcdDevice= 0.05
[  485.523807] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  485.531223] usb 1-1.1: Product: Aoni HD Camera
[  485.535796] usb 1-1.1: Manufacturer: Sunplus IT Co 
[  485.820950] usb 1-1.1: Warning! Unlikely big volume range (=4096), cval->res is probably wrong.
[  485.829869] usb 1-1.1: [5] FU [Mic Capture Volume] ch = 1, val = 0/4096/1
```
This is the dmesg that I can see !  I  can't find the file  be created  about usb camera.

jaeminkang2, 2020-06-02:
Hi 

I don't think that I can see it on dmsg.
I meant logcat. But I don't know if the contents  of file creation will be shown in log.
And.. If you can't see it in logcat, i think that you can also check the location of the file to be created in adb shell.
It just my thought.  Good luck ^^

Midysen, 2020-06-04:
I have solved this problem. Please confirm the  kernel configuration of " USB Video Class (UVC)" is "*".

jaeminkang2, 2020-06-04:
Sorry. I don't know that much.
I just talked about test content.

But I'm glad you solved it. Good luck. ^^

closed 2020-06-07 by peyo-hd