android-rpi/device_arpi_rpi4 #18: Graphics HAL of this build... What is this mean?

jaeminkang2, 2020-06-02:
on the screen of local_manifests's README
------------------------------------------------------------------------------
Graphics HAL of this build : https://gitlab.freedesktop.org/mesa/mesa
------------------------------------------------------------------------------
Graphics HAL of this build... What is this mean?

I thought I should compare this mesa source and android source (/external/mesa).

If anyone knows about this, please explain it to me more specifically.

peyo-hd, 2020-06-02:
It just means mesa is used for OpenGL implementation of this build.
Mesa source (/external/mesa3d) is slightly modified version from gitlab mesa source.

jaeminkang2, 2020-06-02:
Okay.. I understand it .. Thank you

closed 2020-06-02 by jaeminkang2