android-rpi/device_arpi_rpi4 #19: Issue with audio / video

jaeminkang2, 2020-06-03:
Hi arpi team, 

I would like ask about the issue of audio/video.
My current rpi4 status is android 10 (r35) completed a boot success.

I've already seen your inquiry about the same issue, and I'm waiting for it.
However, I'm checking for the audio issue to solve, but it's hard.

I know the questions below are stupid, but...
1. Is it a good choice to compare rpi3 with rpi4 audio part in source?
2. Below is logcat log when mp3 file is operated as vlc player.
------------------------------------------------------------------
06-03 00:02:23.124  1162  2304 E audio_hw_primary: pcm_open(out) failed: cannot open device '/dev/snd/pcmC0D1p': No such file or directory
06-03 00:02:24.772  1162  2304 I chatty  : uid=1041(audioserver) writer identical 157 lines
06-03 00:02:24.782  1162  2304 E audio_hw_primary: pcm_open(out) failed: cannot open device '/dev/snd/pcmC0D1p': No such file or directory
06-03 00:02:24.793  1647  2300 W VLC     : [8038fab0/8fc] libvlc audio output: playback too late (65145): up-sampling
06-03 00:02:24.793  1162  2304 E audio_hw_primary: pcm_open(out) failed: cannot open device '/dev/snd/pcmC0D1p': No such file or directory
06-03 00:02:24.804  1162  2304 I chatty  : uid=1041(audioserver) writer identical 1 line
06-03 00:02:24.814  1162  2304 E audio_hw_primary: pcm_open(out) failed: cannot open device '/dev/snd/pcmC0D1p': No such file or directory
06-03 00:02:24.820  1647  2300 D VLC     : [8061aeb0/8fc] libvlc audio resampler: Using 'vr32' engine
06-03 00:02:24.825  1162  2304 E audio_hw_primary: pcm_open(out) failed: cannot open device '/dev/snd/pcmC0D1p': No such file or directory

jaeminkang2, 2020-06-04:
Hi.. 

I solved the issue. I changed to pcm_card, pcm_device as shown in /dev/snd.
The framework volume api are not work, but audio is output.

If I will solve the volume issue, I will close this issue .

jaeminkang2, 2020-06-08:
Hi arpi team, 

I have additional questions.
When I modified it to PCM_CARD 1 / PCM_DEVICE 0, the sound was output on the earphone.
However, It do output too small sound  and it can not control the volume.

Today I checked your "hardware_arpi/audio" commit.
After modifying to PCM_CARD 0, PCM_DEVICE 0, no sound is output from the audio_jack (earphone).
I would like to ask if additional modifications are needed.
Can you explain a little about the related modifications?

peyo-hd, 2020-06-08:
sync kernel/arpi/ again, since it's updated to 5.4.44

jaeminkang2, 2020-06-09:
I can hear the audio well.
However, the volume is not working yet. ^^;

peyo-hd, 2020-06-16:
The volume control issue is fixed.
Try sync kernel/arpi agina, it's now 5.4.45.

Fixed point is bcm2711_defconfig : CONFIG_ION_SYSTEM_HEAP=y

jaeminkang2, 2020-06-16:
Thank you for your update.
I will redownload source and check it.
Thank you again for your support and update.

I have a add question.
I tried to play mp4 on a regular app using media player.
However, log like below is shown and does not play.

dmesg log 
[  288.094390] audit: rate limit exceeded
[  288.118468] type=1400 audit(1592261775.197:147): avc: denied { getattr } for comm="RenderThread" path="/vendor/lib/hw/gralloc.rpi4.so" dev="mmcblk0p3" ino=277 scontext=u:r:system_server:s0 tcontext=u:object_r:vendor_file:s0 tclass=file permissive=1
[  288.143123] type=1400 audit(1592261775.197:148): avc: denied { read } for comm="RenderThread" name="gralloc.rpi4.so" dev="mmcblk0p3" ino=277 scontext=u:r:system_server:s0 tcontext=u:object_r:vendor_file:s0 tclass=file permissive=1
[  288.167722] type=1400 audit(1592261775.197:148): avc: denied { read } for comm="RenderThread" name="gralloc.rpi4.so" dev="mmcblk0p3" ino=277 scontext=u:r:system_server:s0 tcontext=u:object_r:vendor_file:s0 tclass=file permissive=1
[  288.192625] type=1400 audit(1592261775.197:149): avc: denied { open } for comm="RenderThread" path="/vendor/lib/hw/gralloc.rpi4.so" dev="mmcblk0p3" ino=277 scontext=u:r:system_server:s0 tcontext=u:object_r:vendor_file:s0 tclass=file permissive=1
[  288.217988] type=1400 audit(1592261775.197:149): avc: denied { open } for comm="RenderThread" path="/vendor/lib/hw/gralloc.rpi4.so" dev="mmcblk0p3" ino=277 scontext=u:r:system_server:s0 tcontext=u:object_r:vendor_file:s0 tclass=file permissive=1
[  288.245873] type=1400 audit(1592261775.197:150): avc: denied { read } for comm="RenderThread" name="uevent" dev="sysfs" ino=3559 scontext=u:r:system_server:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[  288.269772] type=1400 audit(1592261775.197:150): avc: denied { read } for comm="RenderThread" name="uevent" dev="sysfs" ino=3559 scontext=u:r:system_server:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[  288.292371] type=1400 audit(1592261775.197:151): avc: denied { open } for comm="RenderThread" path="/sys/devices/platform/v3dbus/fec00000.v3d/uevent" dev="sysfs" ino=3559 scontext=u:r:system_server:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1

I checked the content, and... I'm not sure it is a Selinux issue.
I don't know how to check after that.

peyo-hd, 2020-06-18:
Added guidance for Video decoder in following link : 

https://github.com/android-rpi/device_arpi_rpi4/wiki/Android-10-:-patch-framework-source

jaeminkang2, 2020-06-19:
Thanks for your guidance. I will check it soon.

jaeminkang2, 2020-06-22:
I have one more thing.. 

In your guide... In the SoftwareRender.cpp file...
'OMX_TI_COLOR_FormatYUV420Planar' not found.

Do you mean 'OMX_COLOR_FormatYUV420Planar'?

peyo-hd, 2020-06-22:
'OMX_TI_COLOR_FormatYUV420Planar' was typo.

I have corrected to 'OMX_COLOR_FormatYUV420Planar'.

Thanks for finding it.

jaeminkang2, 2020-06-23:
Hi
I solved most of my media related issues in the latest git yesterday.
And also the my app using media player was also normal output.
Thanks again peyo & arpi team.

Bauuuuu, 2020-08-10:
something more changed? PCM_CARD/PCM_DEVICE didn't had any effect? i tried all combinations and still had audio in audio jack. I was hoping for hdmi audio

Bauuuuu, 2020-08-11:
> something more changed? PCM_CARD/PCM_DEVICE didn't have any effect? i tried all combinations and still had audio in audio jack. I was hoping for hdmi audio

@peyo-hd any hints?

hoangthanh49, 2020-08-12:
hi Bau
Did you try to set hdmi_drive=2 on config.txt?
How about test result?

Bauuuuu, 2020-08-12:
@hoangthanh49 indeed i tried changes in config.txt too
 hdmi_drive=2 and hdmi_force_edid_audio=1 and several other combinations although the result was same. No HDMI audio bu in audio-jack

jaeminkang2, 2020-08-19:
Hello 
Is it possible to print audio with hdmi?
May I ask for support for this part?

closed 2020-09-22 by jaeminkang2