android-rpi/device_arpi_rpi4 #24: [Question] Successful boot, no home or back buttons

kaushal98, 2020-06-16:
Hi android-rpi team -

I was succesfully able to build and boot RPi4 8 GB from this repo. However, when the system boots there are no system buttons and all the UI has is just 2 apps  -`System Tracing` and `Settings`.

Can someone please advice if this is expected? Or somehow my build or boot is not right?

Here's a screenshot after the system has booted.

[image]

jaeminkang2, 2020-06-16:
Hi 
Your screen is normal.. It just RpLauncher app. 
If you expacted other screen, you can install other Launcher app.

hkoosha, 2020-06-17:
The default build is a TV device plus a custom launcher. To have the navbar + statusbar, first make it a tablet then remove the RPLaucnher and use the default launcher. As in here:
https://github.com/hkoosha/device_arpi_rpi4

RaspberryPiFan, 2020-06-17:
To use that repo, you need to have https://gitlab.com/openrpi/rpidroid/vendor/openrpi/taskbar and https://gitlab.com/openrpi/rpidroid/vendor/openrpi/calculator in your AOSP source tree.

kaushal98, 2020-06-17:
Thanks everyone. I didn't realize from the ReadMe that this repo builds Android TV. Even though the patch does mention `TvSettings`, I didn't think it was actually TV.

kaushal98, 2020-06-17:
> The default build is a TV device plus a custom launcher. To have the navbar + statusbar, first make it a tablet then remove the RPLaucnher and use the default launcher. As in here:
> https://github.com/hkoosha/device_arpi_rpi4

Hi @hkoosha since there is no readme in that repo, are the build instructions same as in this one? Sorry newbie here :)

RaspberryPiFan, 2020-06-17:
They are mostly the same, but you need to run `lunch erpi4-eng` instead of `lunch rpi4-eng`

jylee256, 2020-06-23:
Hi @hkoosha and everyone
Since I applied your new guide of tablet default launcher on this issue "https://github.com/hkoosha/device_arpi_rpi4", my android isn't booted printing the gatekeeper abort() log and restarting the init process.
--------------------------------
447 [    7.002971] DEBUG: pid: 261, tid: 261, name: android.hardwar  >>> /vendor/bin/hw/android.hardware.gatekeeper@1.0-service <<<
448 [    7.008495] DEBUG: uid: 1000
449 [    7.020413] DEBUG: signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
450 [    7.023670] DEBUG: Abort message: 'Unable to open GateKeeper HAL'
451 [    7.031647] DEBUG:     r0  00000000  r1  00000105  r2  00000006  r3  bec46030
452 [    7.038013] DEBUG:     r4  bec46044  r5  bec46028  r6  00000105  r7  0000016b
453 [    7.046148] DEBUG:     r8  bec46040  r9  bec46030  r10 bec46060  r11 bec46050
454 console:/ $ [    7.053651] DEBUG:     ip  00000105  sp  bec46000  lr  a9ce62a3  pc  a9ce62b6
455 [    7.179251] DEBUG:
456 [    7.198204] DEBUG: backtrace:
457 [    7.198369] DEBUG:       #00 pc 0005f2b6  /apex/com.android.runtime/lib/bionic/libc.so (abort+166) (BuildId: d1b50ae452ed4e239b3fc339ada39770)
458 [    7.201529] DEBUG:       #01 pc 00005a31  /system/lib/liblog.so (__android_log_assert+176) (BuildId: 5308d97a8ae4abb015fa115d4dabe0b0)
459 [    7.215206] DEBUG:       #02 pc 000025f7  /vendor/lib/hw/android.hardware.gatekeeper@1.0-impl.so (HIDL_FETCH_IGatekeeper+94) (BuildId: 1833422b1c0f5a7e46a98ca931517328)
460 [    7.228004] DEBUG:       #03 pc 000418d3  /system/lib/vndk-sp-29/libhidlbase.so (_ZZN7android8hardware25PassthroughServiceManager3getERKNS0_11hidl_stringES4_ENKUlPvRKNSt3__112basic_stringIcNS6_11char_)
461 [    7.237126] *** NOT YET: opcode db ***
462 [    7.243960] DEBUG:       #04 pc 0003e9e3  /system/lib/vndk-sp-29/libhidlbase.so (android::hardware::PassthroughServiceManager::openLibs(std::__1::basic_string<char, std::__1::char_traits<char>, std::_)
463 [    7.244087] DEBUG:       #05 pc 00040737  /system/lib/vndk-sp-29/libhidlbase.so (android::hardware::PassthroughServiceManager::get(android::hardware::hidl_string const&, android::hardware::hidl_string)
464 [    7.318806] DEBUG:       #06 pc 0003f659  /system/lib/vndk-sp-29/libhidlbase.so (android::hardware::details::getRawServiceInternal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::a)
465 [    7.342094] DEBUG:       #07 pc 0000b9bf  /system/lib/vndk-29/android.hardware.gatekeeper@1.0.so (_ZN7android8hardware7details18getServiceInternalINS0_10gatekeeper4V1_014BpHwGatekeeperENS4_11IGatekeep)
466 [    7.376200] DEBUG:       #08 pc 0000bad7  /system/lib/vndk-29/android.hardware.gatekeeper@1.0.so (android::hardware::gatekeeper::V1_0::IGatekeeper::getService(std::__1::basic_string<char, std::__1::ch)
467 [    7.406413] DEBUG:       #09 pc 000010cf  /vendor/bin/hw/android.hardware.gatekeeper@1.0-service (BuildId: 721eaa8b1df6bad1c941acfe337e1d03)
468 [    7.433911] DEBUG:       #10 pc 00001085  /vendor/bin/hw/android.hardware.gatekeeper@1.0-service (main+64) (BuildId: 721eaa8b1df6bad1c941acfe337e1d03)
469 [    7.446862] DEBUG:       #11 pc 000598f9  /apex/com.android.runtime/lib/bionic/libc.so (__libc_init+68) (BuildId: d1b50ae452ed4e239b3fc339ada39770)
470 [    7.460686] DEBUG:       #12 pc 0000102f  /vendor/bin/hw/android.hardware.gatekeeper@1.0-service (_start_main+38) (BuildId: 721eaa8b1df6bad1c941acfe337e1d03)
471 [    7.474541] DEBUG:       #13 pc 00004456  <anonymous:aa7de000>
472 [    9.899231] init: Received control message 'interface_start' for 'android.hardware.gatekeeper@1.0::IGatekeeper/default' from pid: 208 (/system/bin/hwservicemanager)
473 [    9.920275] init: Could not find 'android.hardware.gatekeeper@1.0::IGatekeeper/default' for ctl.interface_start
----------------------------------------------------------------------------

I followed the README guide of this repo, and then I just modified the default.xml file for applying the new repo "https://github.com/hkoosha/device_arpi_rpi4", like below

<default.xml>
  2 <manifest>
  3   <remote name="arpi" fetch="https://github.com/android-rpi"/>
  **4   <remote name="hkoosha" fetch="https://github.com/hkoosha"/>**
  5
  6   <remove-project name="platform/external/mesa3d"/>
  7   <project path="external/mesa3d" name="external_mesa3d" revision="v3d-20.1" remote="arpi"/>
  8   <project path="external/drm_gralloc" name="external_drm_gralloc" revision="v3d-20.1" remote="arpi"/>
  9
 10   <project path="kernel/arpi" name="kernel_arpi" revision="arpi-5.4.y" remote="arpi"/>
 11   <project path="hardware/arpi" name="hardware_arpi" revision="arpi-10" remote="arpi"/>
 **12   <project path="device/endorphin/erpi4" name="device_arpi_rpi4" revision="arpi-10" remote="hkoosha"/>**

Is there anything that I missed??
I would really appreciate it if you help me.

jylee256, 2020-06-23:
Hi @hkoosha and everyone
Since I applied your new guide of a tablet and default launcher "https://github.com/hkoosha/device_arpi_rpi4",
my android isn't booted printing the gatekeeper abort() log and restarting init process.

[    6.972683] DEBUG: Build fingerprint: 'endorphin/erpi4/erpi4:10/QQ3A.200605.002.A1/eng.jylee2.20200622.102123:eng/test-keys'
[    6.984360] DEBUG: Revision: '0'
[    6.996401] DEBUG: ABI: 'arm'
[    6.999878] DEBUG: Timestamp: 1970-01-01 00:00:06+0000
[    7.002971] DEBUG: pid: 261, tid: 261, name: android.hardwar  >>> /vendor/bin/hw/android.hardware.gatekeeper@1.0-service <<<
[    7.008495] DEBUG: uid: 1000
[    7.020413] DEBUG: signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
[    7.023670] DEBUG: Abort message: 'Unable to open GateKeeper HAL'
[    7.031647] DEBUG:     r0  00000000  r1  00000105  r2  00000006  r3  bec46030
[    7.038013] DEBUG:     r4  bec46044  r5  bec46028  r6  00000105  r7  0000016b
[    7.046148] DEBUG:     r8  bec46040  r9  bec46030  r10 bec46060  r11 bec46050
[    7.053651] DEBUG:     ip  00000105  sp  bec46000  lr  a9ce62a3  pc  a9ce62b6
[    7.179251] DEBUG:
[    7.198204] DEBUG: backtrace:
[    7.198369] DEBUG:       #00 pc 0005f2b6  /apex/com.android.runtime/lib/bionic/libc.so (abort+166) (BuildId: d1b50ae452ed4e239b3fc339ada39770)
[    7.201529] DEBUG:       #01 pc 00005a31  /system/lib/liblog.so (__android_log_assert+176) (BuildId: 5308d97a8ae4abb015fa115d4dabe0b0)
[    7.215206] DEBUG:       #02 pc 000025f7  /vendor/lib/hw/android.hardware.gatekeeper@1.0-impl.so (HIDL_FETCH_IGatekeeper+94) (BuildId: 1833422b1c0f5a7e46a98ca931517328)
[    7.228004] DEBUG:       #03 pc 000418d3  /system/lib/vndk-sp-29/libhidlbase.so (_ZZN7android8hardware25PassthroughServiceManager3getERKNS0_11hidl_stringES4_ENKUlPvRKNSt3__112basic_stringIcNS6_11char_)
[    7.237126] *** NOT YET: opcode db ***
[    7.243960] DEBUG:       #04 pc 0003e9e3  /system/lib/vndk-sp-29/libhidlbase.so (android::hardware::PassthroughServiceManager::openLibs(std::__1::basic_string<char, std::__1::char_traits<char>, std::_)
[    7.244087] DEBUG:       #05 pc 00040737  /system/lib/vndk-sp-29/libhidlbase.so (android::hardware::PassthroughServiceManager::get(android::hardware::hidl_string const&, android::hardware::hidl_string)
[    7.318806] DEBUG:       #06 pc 0003f659  /system/lib/vndk-sp-29/libhidlbase.so (android::hardware::details::getRawServiceInternal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::a)
[    7.342094] DEBUG:       #07 pc 0000b9bf  /system/lib/vndk-29/android.hardware.gatekeeper@1.0.so (_ZN7android8hardware7details18getServiceInternalINS0_10gatekeeper4V1_014BpHwGatekeeperENS4_11IGatekeep)
[    7.376200] DEBUG:       #08 pc 0000bad7  /system/lib/vndk-29/android.hardware.gatekeeper@1.0.so (android::hardware::gatekeeper::V1_0::IGatekeeper::getService(std::__1::basic_string<char, std::__1::ch)
[    7.406413] DEBUG:       #09 pc 000010cf  /vendor/bin/hw/android.hardware.gatekeeper@1.0-service (BuildId: 721eaa8b1df6bad1c941acfe337e1d03)
[    7.433911] DEBUG:       #10 pc 00001085  /vendor/bin/hw/android.hardware.gatekeeper@1.0-service (main+64) (BuildId: 721eaa8b1df6bad1c941acfe337e1d03)
[    7.446862] DEBUG:       #11 pc 000598f9  /apex/com.android.runtime/lib/bionic/libc.so (__libc_init+68) (BuildId: d1b50ae452ed4e239b3fc339ada39770)
[    7.460686] DEBUG:       #12 pc 0000102f  /vendor/bin/hw/android.hardware.gatekeeper@1.0-service (_start_main+38) (BuildId: 721eaa8b1df6bad1c941acfe337e1d03)
[    7.474541] DEBUG:       #13 pc 00004456  <anonymous:aa7de000>
[    9.899231] init: Received control message 'interface_start' for 'android.hardware.gatekeeper@1.0::IGatekeeper/default' from pid: 208 (/system/bin/hwservicemanager)
[    9.920275] init: Could not find 'android.hardware.gatekeeper@1.0::IGatekeeper/default' for ctl.interface_start
[   10.900686] init: Received control message 'interface_start' for 'android.hardware.gatekeeper@1.0::IGatekeeper/default' from pid: 208 (/system/bin/hwservicemanager)
[   10.915727] init: Could not find 'android.hardware.gatekeeper@1.0::IGatekeeper/default' for ctl.interface_start
[   11.731213] init: starting service 'netd'...
[   11.741378] init: starting service 'zygote'...

I followed your README guide on this repo, and then I just modified the default.xml for applying new endorphin repo, like below,

<default.xml>
< remote name="arpi" fetch="https://github.com/android-rpi"/>
**< remote name="hkoosha" fetch="https://github.com/hkoosha"/>**
 
< remove-project name="platform/external/mesa3d"/>
< project path="external/mesa3d" name="external_mesa3d" revision="v3d-20.1" remote="arpi"/>
< project path="external/drm_gralloc" name="external_drm_gralloc" revision="v3d-20.1" remote="arpi"/>
  
< project path="kernel/arpi" name="kernel_arpi" revision="arpi-5.4.y" remote="arpi"/>
< project path="hardware/arpi" name="hardware_arpi" revision="arpi-10" remote="arpi"/>
**< project path="device/endorphin/erpi4" name="device_arpi_rpi4" revision="arpi-10" remote="hkoosha"/>**
 
Is there anything that I missed?
I would really appreciate it if you help me, Thanks

kaushal98, 2020-06-24:
> They are mostly the same, but you need to run `lunch erpi4-eng` instead of `lunch rpi4-eng`

Hi @RaspberryPiFan @hkoosha is this correct build step though - 

```

kaushal98, 2020-06-24:
@jylee256 I'm doing the same thing, just started building so will take a while. Here's what my `default.xml` looks like

```
<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote name="arpi" fetch="https://github.com/android-rpi"/>
  <remote name="erpi" fetch="https://github.com/hkoosha"/>
  
  <remove-project name="platform/external/mesa3d"/>
  <project path="external/mesa3d" name="external_mesa3d" revision="v3d-20.1" remote="arpi"/>
  <project path="external/drm_gralloc" name="external_drm_gralloc" revision="v3d-20.1" remote="arpi"/>

  <project path="kernel/arpi" name="kernel_arpi" revision="arpi-5.4.y" remote="arpi"/>
  <project path="hardware/arpi" name="hardware_arpi" revision="arpi-10" remote="arpi"/>
  <project path="device/arpi/rpi4" name="device_arpi_rpi4" revision="arpi-10" remote="arpi"/>
  <project path="device/endorphin/erpi4" name="device_arpi_rpi4" revision="arpi-10" remote="erpi"/>

  <!-- 
   <project path="vendor/arpi/RpLauncher" name="RpLauncher" revision="arpi-10" remote="arpi"/>
  -->
  <remove-project name="device/amlogic/yukawa-kernel" />
  <remove-project name="device/google/bonito-kernel" />
  <remove-project name="device/google/coral-kernel" />
  <remove-project name="device/google/crosshatch-kernel" />
  <remove-project name="device/google/marlin-kernel" />
  <remove-project name="device/google/wahoo-kernel" />
  <remove-project name="device/linaro/hikey-kernel" />
  <remove-project name="device/linaro/poplar-kernel" />
  <remove-project name="device/ti/beagle-x15-kernel" />
</manifest>
```

kaushal98, 2020-07-09:
closing for now since original issue is resolved

closed 2020-07-09 by kaushal98