android-rpi/device_arpi_rpi4 #25: Bluetooth

ramasamyesvar, 2020-06-19:
Thanks all for the Nice work. Is  Bluetooth enabled in this?.  I am not able to find it in TV image. 
 I am trying with the camera patch from RaspberryPiFan. I would like to include Bluetooth changes if there are some. Thanks again

ramasamyesvar, 2020-06-19:
Can I port this with the TV setup https://github.com/hkoosha/device_arpi_rpi4/pull/1/commits/c3cf22a429f3d6e5b6fd328999f5904a19a52870 ?

hkoosha, 2020-06-19:
well I tried that but it didn't work. the image won't boot anymore. you could go ahead and try maybe you'll have better luck .

RaspberryPiFan, 2020-06-19:
Bluetooth support is at https://github.com/peyo-hd/device_arpi_rpi4/tree/bt_0607

ramasamyesvar, 2020-06-20:
Yes, I took the same changes manually and the "Searching for Accessories" screen waits for long and does not find any Bluetooth devices.  Pretty sure I didn't miss anything. Does the log  below 
[bluetoothlog.txt](https://github.com/android-rpi/device_arpi_rpi4/files/4807381/bluetoothlog.txt)

tell anything?

(BuildId: d1b50ae452ed4e239b3fc339ada39770)
2020-06-20 10:37:07.380 384-726/system_process E/WifiVendorHal: getWifiLinkLayerStats_1_3_Internal(l.926) failed {.code = ERROR_NOT_AVAILABLE, .description = }
2020-06-20 10:37:07.691 32753-366/com.android.bluetooth E/AdapterState: BLE_TURNING_ON : BLE_START_TIMEOUT
2020-06-20 10:37:07.691 32753-366/com.android.bluetooth I/AdapterState: BLE_TURNING_OFF : entered 
2020-06-20 10:37:07.691 32753-366/com.android.bluetooth D/AdapterProperties: Setting state to BLE_TURNING_OFF
2020-06-20 10:37:07.691 32753-366/com.android.bluetooth D/BluetoothAdapterService: updateAdapterState() - Broadcasting state BLE_TURNING_OFF to 1 receivers.
2020-06-20 10:37:07.692 384-405/system_process D/BluetoothManagerService: MESSAGE_BLUETOOTH_STATE_CHANGE: BLE_TURNING_ON > BLE_TURNING_OFF
2020-06-20 10:37:07.692 384-405/system_process D/BluetoothManagerService: Sending BLE State Change: BLE_TURNING_ON > BLE_TURNING_OFF
2020-06-20 10:37:07.692 32753-366/com.android.bluetooth D/AdapterProperties: onBleDisable
2020-06-20 10:37:07.695 32753-32753/com.android.bluetooth D/BtGatt.DebugUtils: handleDebugAction() action=null
2020-06-20 10:37:07.702 32753-32753/com.android.bluetooth W/BtGatt.ScanManager: exception when invoking removeOnUidImportanceListener

Looks like it Crashes here

2020-06-20 10:37:03.721 32753-367/? I/bt_stack_manager: event_start_up_stack is bringing up the stack
2020-06-20 10:37:03.721 32753-367/? I/bt_core_module: module_start_up Starting module "btif_config_module"
2020-06-20 10:37:03.721 32753-367/? I/bt_core_module: module_start_up Started module "btif_config_module"
2020-06-20 10:37:03.721 32753-367/? I/bt_core_module: module_start_up Starting module "btsnoop_module"
2020-06-20 10:37:03.722 32753-367/? I/bt_core_module: module_start_up Started module "btsnoop_module"
2020-06-20 10:37:03.722 32753-367/? I/bt_core_module: module_start_up Starting module "hci_module"
2020-06-20 10:37:03.722 32753-367/? I/bt_hci: hci_module_start_up
2020-06-20 10:37:03.722 32753-367/? D/bt_hci: hci_module_start_up starting async portion
2020-06-20 10:37:03.722 32753-431/? I/bt_hci: hci_initialize
2020-06-20 10:37:03.724 32753-431/? I/bt_hci: hci_initialize: IBluetoothHci::getService() returned 0x92bfe5a0 (remote)
2020-06-20 10:37:03.724 255-273/? I/android.hardware.bluetooth@1.0-impl: BluetoothHci::initialize()
2020-06-20 10:37:03.725 255-273/? I/bt_vendor: init
2020-06-20 10:37:03.725 255-273/? I/bt_vnd_conf: Attempt to load conf from /etc/bluetooth/bt_vendor.conf
2020-06-20 10:37:03.725 255-273/? D/android.hardware.bluetooth@1.0-impl: Open vendor library loaded
2020-06-20 10:37:03.725 255-273/? W/bt_vendor: NOTE: BT_VND_PWR_ON now forces power-off first
2020-06-20 10:37:03.725 255-273/? I/bt_userial_vendor: userial vendor open: opening /dev/ttyAMA0
2020-06-20 10:37:03.726 255-273/? I/bt_userial_vendor: device fd = 6 open
2020-06-20 10:37:04.366 384-726/system_process E/WifiVendorHal: getWifiLinkLayerStats_1_3_Internal(l.926) failed {.code = ERROR_NOT_AVAILABLE, .description = }
2020-06-20 10:37:04.475 384-397/system_process I/system_server: Background concurrent copying GC freed 19128(1301KB) AllocSpace objects, 31(3580KB) LOS objects, 42% free, 5765KB/10089KB, paused 138us total 111.407ms
2020-06-20 10:37:06.622 32753-380/com.android.bluetooth E/bt_hci: startup_timer_expired
2020-06-20 10:37:06.622 32753-380/com.android.bluetooth A/libc: Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 380 (alarm_default_c), pid 32753 (droid.bluetooth)
2020-06-20 10:37:06.699 463-463/? I/crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
2020-06-20 10:37:06.700 298-298/? I//system/bin/tombstoned: received crash request for pid 380
2020-06-20 10:37:06.700 463-463/? I/crash_dump32: performing dump of process 32753 (target tid = 380)
2020-06-20 10:37:06.716 463-463/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***

ramasamyesvar, 2020-06-20:
I missed this  https://github.com/peyo-hd/device_arpi_rpi4/pull/1/commits/80544983304266e1f9f359a2b352a811521912fa. Will try and update

RaspberryPiFan, 2020-06-20:
That probably won't help. I'll try to find a fix. Your logfile helps a lot. Thank you!

RaspberryPiFan, 2020-06-20:
Try to apply this: https://github.com/peyo-hd/device_arpi_rpi4/commit/ead4242f783940c8697f6fb43677f727d39c50ae

RaspberryPiFan, 2020-06-20:
What repo are you using to build? Please don't use hkoosha's repo, it could have to do with the rename in this repo.

ramasamyesvar, 2020-06-20:
I was using arpi-10 from android-rpi  + your camera fix and + bluetooth fix.
I noticed your camera fix has Conflicting files
boot/fixup4.dat
boot/start4.elf
from camera patch which may not working with 5.4.47.  Please check that. That may be one issue in my previous build. 

I have now forked arpi-10 tip and applied https://github.com/peyo-hd/device_arpi_rpi4/tree/bt_0607 alone now. will update on this tomorrow with fresh build

RaspberryPiFan, 2020-06-20:
Thank you! I fixed the PR.

ramasamyesvar, 2020-06-21:
Still The bluetooth asserts and is not working . I using the lastest arpi-10 +  https://github.com/peyo-hd/device_arpi_rpi4/tree/bt_0607. 

2020-06-21 23:00:25.922 9590-9590/? I/Zygote: seccomp disabled by setenforce 0
2020-06-21 23:00:26.061 9590-9590/com.android.bluetooth D/BluetoothOppFileProvider: Initialized
2020-06-21 23:00:26.080 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding A2dpService
2020-06-21 23:00:26.080 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding HidHostService
2020-06-21 23:00:26.081 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding PanService
2020-06-21 23:00:26.081 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding GattService
2020-06-21 23:00:26.081 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding BluetoothMapService
2020-06-21 23:00:26.081 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding AvrcpTargetService
2020-06-21 23:00:26.081 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding HidDeviceService
2020-06-21 23:00:26.081 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding BluetoothOppService
2020-06-21 23:00:26.081 9590-9590/com.android.bluetooth V/AdapterServiceConfig: Adding BluetoothPbapService
2020-06-21 23:00:26.112 9590-9590/com.android.bluetooth D/BluetoothAdapterService: onCreate()
2020-06-21 23:00:26.122 9590-9590/com.android.bluetooth D/AdapterState: make() - Creating AdapterState
2020-06-21 23:00:26.124 9590-9621/com.android.bluetooth I/AdapterState: OFF : entered 
2020-06-21 23:00:26.124 9590-9621/com.android.bluetooth D/AdapterProperties: Setting state to OFF
2020-06-21 23:00:26.126 9590-9590/com.android.bluetooth I/bt_btif: init: start restricted = 0 ; niap = 0
2020-06-21 23:00:26.126 9590-9590/com.android.bluetooth D/bt_osi_allocation_tracker: canary initialized
2020-06-21 23:00:26.126 9590-9622/com.android.bluetooth I/bt_stack_manager: event_init_stack is initializing the stack
2020-06-21 23:00:26.127 9590-9622/com.android.bluetooth W/bt_btif_config: init unable to load config file: /data/misc/bluedroid/bt_config.conf; using backup.
2020-06-21 23:00:26.127 9590-9622/com.android.bluetooth W/bt_btif_config: init unable to load backup; attempting to transcode legacy file.
2020-06-21 23:00:26.127 9590-9622/com.android.bluetooth E/bt_btif_config_transcode: btif_config_transcode unable to load XML file '/data/misc/bluedroid/bt_config.xml': 3
2020-06-21 23:00:26.127 9590-9622/com.android.bluetooth E/bt_btif_config: init unable to transcode legacy file; creating empty config.
2020-06-21 23:00:26.128 9590-9622/com.android.bluetooth E/bt_osi_alarm: timer_create_internal unable to create timer with clock 9: Operation not supported on transport endpoint
2020-06-21 23:00:26.128 9590-9622/com.android.bluetooth E/bt_osi_alarm: The kernel might not have support for timer_create(CLOCK_BOOTTIME_ALARM): https://lwn.net/Articles/429925/
2020-06-21 23:00:26.128 9590-9622/com.android.bluetooth E/bt_osi_alarm: See following patches: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/log/?qt=grep&q=CLOCK_BOOTTIME_ALARM
2020-06-21 23:00:26.128 9590-9626/com.android.bluetooth I/bt_osi_thread: run_thread: thread id 9626, thread name alarm_default_ca started
2020-06-21 23:00:26.129 9590-9627/com.android.bluetooth I/bt_osi_thread: run_thread: thread id 9627, thread name alarm_dispatcher started
2020-06-21 23:00:26.131 9590-9622/com.android.bluetooth I/bt_btif_core: btif_init_bluetooth entered
2020-06-21 23:00:26.131 9590-9622/com.android.bluetooth I/bt_stack_config: init attempt to load stack conf from /etc/bluetooth/bt_stack.conf
2020-06-21 23:00:26.132 9590-9622/com.android.bluetooth I/bt_btif_core: btif_init_bluetooth finished
2020-06-21 23:00:26.132 9590-9622/com.android.bluetooth I/bt_stack_manager: event_init_stack finished
2020-06-21 23:00:26.132 9590-9590/com.android.bluetooth I/bt_osi_wakelock: wakelock_set_os_callouts set to non-native
2020-06-21 23:00:26.132 9590-9590/com.android.bluetooth I/bt_btif: get_profile_interface: id = socket
2020-06-21 23:00:26.135 9590-9628/com.android.bluetooth E/bt_btif_storage: btif_storage_get_adapter_property: Controller not ready! Unable to return Bluetooth Address
2020-06-21 23:00:26.135 9590-9628/com.android.bluetooth E/BluetoothServiceJni: adapter_properties_callback: Status 1 is incorrect
2020-06-21 23:00:26.135 9590-9590/com.android.bluetooth I/bt_btif: get_profile_interface: id = sdp
2020-06-21 23:00:26.136 9590-9628/com.android.bluetooth D/AdapterProperties: Name is: RPI4
2020-06-21 23:00:26.136 9590-9628/com.android.bluetooth D/AdapterProperties: BT Class:1a0120
2020-06-21 23:00:26.138 9590-9590/com.android.bluetooth I/BluetoothAdapterService: Phone policy enabled
2020-06-21 23:00:26.142 9590-9590/com.android.bluetooth D/BluetoothActiveDeviceManager: start()
2020-06-21 23:00:26.146 9590-9629/com.android.bluetooth D/BluetoothActiveDeviceManager: onAudioDevicesAdded
2020-06-21 23:00:26.146 9590-9629/com.android.bluetooth D/BluetoothActiveDeviceManager: Audio device added: Raspberry Pi 4 type: 2
2020-06-21 23:00:26.146 9590-9629/com.android.bluetooth D/BluetoothActiveDeviceManager: Audio device added: USB-Audio - USB PnP Sound Device type: 11
2020-06-21 23:00:26.161 9590-9590/com.android.bluetooth D/BluetoothDatabase: start()
2020-06-21 23:00:26.162 9590-9590/com.android.bluetooth D/BluetoothDatabase: Load Database
[... cut ...]
2020-06-21 23:00:26.220 9590-9622/com.android.bluetooth D/bt_hci: hci_module_start_up starting async portion
2020-06-21 23:00:26.220 9590-9640/com.android.bluetooth I/bt_hci: hci_initialize
2020-06-21 23:00:26.222 9590-9640/com.android.bluetooth I/bt_hci: hci_initialize: IBluetoothHci::getService() returned 0xababe1e0 (remote)
2020-06-21 23:00:29.120 9590-9626/com.android.bluetooth E/bt_hci: startup_timer_expired
2020-06-21 23:00:29.120 9590-9626/com.android.bluetooth A/libc: Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 9626 (alarm_default_c), pid 9590 (droid.bluetooth)

RaspberryPiFan, 2020-06-22:
He has updated it: https://github.com/peyo-hd/device_arpi_rpi4/tree/bt_0620.

ramasamyesvar, 2020-06-23:
I checkout his branch and tried it alone with out any other changes. I  still see the crash.
Am I missing something on the kernel side. I took the latest from kernel branch.

? I: seccomp disabled by setenforce 0
com.android.bluetooth D: Initialized
com.android.bluetooth V: Adding A2dpService
com.android.bluetooth V: Adding HidHostService
com.android.bluetooth V: Adding PanService
com.android.bluetooth V: Adding GattService
com.android.bluetooth V: Adding BluetoothMapService
com.android.bluetooth V: Adding AvrcpTargetService
com.android.bluetooth V: Adding HidDeviceService
com.android.bluetooth V: Adding BluetoothOppService
com.android.bluetooth V: Adding BluetoothPbapService
com.android.bluetooth D: onCreate()
com.android.bluetooth D: make() - Creating AdapterState
com.android.bluetooth I: OFF : entered 
com.android.bluetooth D: Setting state to OFF
com.android.bluetooth I: init: start restricted = 0 ; niap = 0
com.android.bluetooth D: canary initialized
com.android.bluetooth I: event_init_stack is initializing the stack
com.android.bluetooth E: timer_create_internal unable to create timer with clock 9: Operation not supported on transport endpoint
com.android.bluetooth E: The kernel might not have support for timer_create(CLOCK_BOOTTIME_ALARM): https://lwn.net/Articles/429925/
com.android.bluetooth E: See following patches: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/log/?qt=grep&q=CLOCK_BOOTTIME_ALARM
com.android.bluetooth I: run_thread: thread id 7409, thread name alarm_default_ca started
com.android.bluetooth I: run_thread: thread id 7410, thread name alarm_dispatcher started
com.android.bluetooth I: btif_init_bluetooth entered
com.android.bluetooth I: init attempt to load stack conf from /etc/bluetooth/bt_stack.conf
com.android.bluetooth I: btif_init_bluetooth finished
com.android.bluetooth I: event_init_stack finished
com.android.bluetooth I: wakelock_set_os_callouts set to non-native
com.android.bluetooth I: get_profile_interface: id = socket
com.android.bluetooth E: btif_storage_get_adapter_property: Controller not ready! Unable to return Bluetooth Address
com.android.bluetooth E: adapter_properties_callback: Status 1 is incorrect
com.android.bluetooth D: Name is: RPI4
com.android.bluetooth D: BT Class:1a0120
com.android.bluetooth I: get_profile_interface: id = sdp
com.android.bluetooth I: Phone policy enabled
com.android.bluetooth D: start()
com.android.bluetooth D: onAudioDevicesAdded
com.android.bluetooth D: Audio device added: Raspberry Pi 4 type: 2
com.android.bluetooth D: Audio device added: USB-Audio - USB PnP Sound Device type: 11
com.android.bluetooth D: start()
com.android.bluetooth D: Load Database
com.android.bluetooth D: setAdapterService() - trying to set service to com.android.bluetooth.btservice.AdapterService@7cbb5ee
com.android.bluetooth D: onBind()
com.android.bluetooth D: onBluetoothServiceUp: com.android.bluetooth.btservice.AdapterService$AdapterServiceBinder@a705eaa
com.android.bluetooth D: enable() - Enable called with quiet mode status =  false
com.android.bluetooth I: BLE_TURNING_ON : entered 
com.android.bluetooth D: Setting state to BLE_TURNING_ON
com.android.bluetooth D: updateAdapterState() - Broadcasting state BLE_TURNING_ON to 1 receivers.
com.android.bluetooth D: bleOnProcessStart()
com.android.bluetooth I: init(), maxConnectedAudioDevices, default=5, propertyOverlayed=5, finalValue=5
com.android.bluetooth D: bleOnProcessStart() - Make Bond State Machine
com.android.bluetooth D: make
com.android.bluetooth I: StableState(): Entering Off State
com.android.bluetooth I: cacheMetadata
com.android.bluetooth V: cacheMetadata: found device LocalStorage
com.android.bluetooth V: cacheMetadata: Database is ready
com.android.bluetooth I: classInitNative(L876): classInitNative: Success!
com.android.bluetooth D: handleDebugAction() action=null
com.android.bluetooth D: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@7cbb5ee
com.android.bluetooth I: get_profile_interface: id = gatt
com.android.bluetooth D: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@7cbb5ee
com.android.bluetooth D: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@7cbb5ee
com.android.bluetooth I: event_start_up_stack is bringing up the stack
com.android.bluetooth I: module_start_up Starting module "btif_config_module"
com.android.bluetooth I: module_start_up Started module "btif_config_module"
com.android.bluetooth I: module_start_up Starting module "btsnoop_module"
com.android.bluetooth I: module_start_up Started module "btsnoop_module"
com.android.bluetooth I: module_start_up Starting module "hci_module"
com.android.bluetooth I: hci_module_start_up
com.android.bluetooth I: hci_initialize
com.android.bluetooth D: hci_module_start_up starting async portion
com.android.bluetooth I: hci_initialize: IBluetoothHci::getService() returned 0x9d6dfde0 (remote)
com.android.bluetooth E: startup_timer_expired
com.android.bluetooth A: Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 7409 (alarm_default_c), pid 7373 (droid.bluetooth)

RaspberryPiFan, 2020-06-23:
I don't think so, but I don't know much about this. Please wait until a final version is here. If I find a solution before that happens, I'll tell you.

ramasamyesvar, 2020-06-28:
> well I tried that but it didn't work. the image won't boot anymore. you could go ahead and try maybe you'll have better luck .

Hi, did you get bluetooth or camera working in your rpi or erpi branch

lohriialo, 2020-07-07:
@ramasamyesvar any luck you got bluetooth working?

prashantsinhpvips, 2020-08-14:
In [bt_0719](https://github.com/peyo-hd/device_arpi_rpi4/tree/bt_0719) branch apply below patch for temp fix.

cd hardware/interfaces/

```
diff --git a/bluetooth/1.0/default/vendor_interface.cc b/bluetooth/1.0/default/vendor_interface.cc
index d56e344e7..51f9c8d5e 100644
--- a/bluetooth/1.0/default/vendor_interface.cc
+++ b/bluetooth/1.0/default/vendor_interface.cc
@@ -342,7 +342,7 @@ void VendorInterface::OnFirmwareConfigured(uint8_t result) {
   lib_interface_->op(BT_VND_OP_GET_LPM_IDLE_TIMEOUT, &lpm_timeout_ms);
   ALOGI("%s: lpm_timeout_ms %d", __func__, lpm_timeout_ms);
 
-  bt_vendor_lpm_mode_t mode = BT_VND_LPM_ENABLE;
+  bt_vendor_lpm_mode_t mode = BT_VND_LPM_DISABLE;
   lib_interface_->op(BT_VND_OP_LPM_SET_MODE, &mode);
 
   ALOGD("%s Calling StartLowPowerWatchdog()", __func__);

```

It worked for me.

peyo-hd, 2020-08-15:
@prashantsinhpvips ,  Thanks for finding workaround.
Modified wiki : https://github.com/android-rpi/device_arpi_rpi4/wiki/Android-10-:-patch-framework-source
Applied the patch : https://github.com/android-rpi/device_arpi_rpi4/commit/909cd48817ba72b0a84f645a98cb8d33b00f983f

lohriialo, 2020-08-18:
@peyo-hd thanks for the changes, latest commit now supports bluetooth, tested and working!

closed 2020-09-23 by peyo-hd