android-rpi/device_arpi_rpi4 #26: Wifi connect function not work

hoangthanh49, 2020-06-25:
Hello arpi team

I have updated latest source code from this github and tried to test Wifi function.
Scan function worked well but connect function didn't work.
Error log:
2020-06-25 11:14:28.514 329-734/system_process I/WifiService: startScan uid=1000
2020-06-25 11:14:28.528 329-416/system_process D/WifiConfigStore: Writing to stores completed in 17 ms.
2020-06-25 11:14:28.529 329-416/system_process D/WifiClientModeImpl: connectToUserSelectNetwork netId 0, uid 1000, forceReconnect = false
2020-06-25 11:14:28.551 329-416/system_process D/WifiConfigStore: Writing to stores completed in 18 ms.
2020-06-25 11:14:28.553 329-416/system_process E/WifiClientModeImpl: CMD_START_CONNECT but no requests and not connected, bailing
2020-06-25 11:14:31.050 329-348/system_process I/EthernetTracker: interfaceLinkStateChanged, iface: wlan0, up: false

ramasamyesvar, 2020-06-25:
Wifi Works fine for me with latest code. Did you add this patch manually https://github.com/android-rpi/device_arpi_rpi4/wiki/Android-10-:-patch-framework-source  too.

hoangthanh49, 2020-06-25:
That FW patch is for video decoder. I have added them.
Wifi connection worked well on your device?

hoangthanh49, 2020-06-25:
I don't have problem with input wifi pass.
The problem is that wifi connection not work.

ramasamyesvar, 2020-06-25:
For me it's working with 2.4Ghz WPA2-PSK [AES]. Not sure if that makes any difference.  5Ghz should also work I believe.
If you share the complete logcat, I guess someone could help

hoangthanh49, 2020-06-26:
Thanks for your feedback. I will do more test on my device.

hoangthanh49, 2020-07-01:
Connect function worked on my last build.
Hence closing this issue.

closed 2020-07-01 by hoangthanh49