android-rpi/device_arpi_rpi4 #28: USB boot support?

lohriialo, 2020-06-30:
I've updated my rpi4 firmware to the latest stable(June 15th) which now has official support for USB boot. I'm able to USB boot Raspberry Pi OS for example but when I attempt to boot android-rpi, I can see that rainbow screen but it boot loop. Any help/pointers on how to get it to work?

AaronDewes, 2020-07-02:
Try to use this as fstab.rpi4 and rebuild:

```
#<src>                  <mnt_point>         <type>    <mnt_flags and options>                               <fs_mgr_flags>

/dev/block/sda2    /system             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/sda3    /vendor             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/sda4    /data               ext4      noatime,nosuid,nodev,nomblk_io_submit,errors=panic    wait
```

lohriialo, 2020-07-03:
@AaronDewes thanks, works perfectly with USB boot but won't boot in sdcard mode, not sure if there's a good way to have both at the same time

AaronDewes, 2020-07-03:
There is, but I would need you to run `find /dev` in adb shell and send me the output. If you don't want to use adb, you can also rebuild with `PRODUCT_PACKAGES += Terminal` in rpi4.mk

AaronDewes, 2020-07-03:
Please try this fstab:
```
#<src>                  <mnt_point>         <type>    <mnt_flags and options>                               <fs_mgr_flags>

/dev/block/by-name/system    /system             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/by-name/vendor    /vendor             ext4      ro,barrier=1                                          wait,first_stage_mount
/dev/block/by-name/userdata    /data               ext4      noatime,nosuid,nodev,nomblk_io_submit,errors=panic    wait,first_stage_mount
```
This may result in a bootloop, but please try it. If it does, run `find /dev` with adb shell on an older build. Please make sure the partition names are correct.

AaronDewes, 2020-07-23:
@lohriialo Any progress?

lohriialo, 2020-07-23:
I haven't had a chance to rebuild and test it yet, for now I'm sticking with USB boot as I already have some other stuffs going on in the current build

Road-Drum, 2020-07-23:
@AaronDewes I could probably test this. Just patch the fstab in latest arpi-10 source right?

AaronDewes, 2020-07-23:
Yes, Thank you!

Road-Drum, 2020-07-23:
I'll try building tonight, natively compile on my pi 4 lol

AaronDewes, 2020-07-23:
I don't think that will work, android uses x64 prebuilts while building

Road-Drum, 2020-07-23:
> I don't think that will work, android uses x64 prebuilts while building

could I override those prebuilts and recompile them from scratch? I recall there's an option to not use prebuilts and compile from scratch

AaronDewes, 2020-07-23:
I think you can, but I don't know.

Road-Drum, 2020-07-24:
so far the current script is to build `arm` and on my Pi 4, it's currently running Manjaro ARM which I need to get gcc-arm since Manjaro ARM is shipped with `arm64` arch by default. I'm just toying around so I won't delve too deep, but `ARCH=arm make zImage` partially successful since gcc-arm64 doesn't recognize some gcc-arm specific options

maybe I'll have more success on RPi OS, will try to spin a kvm or lxd to run arm version of that, though I'm not sure how to setup kvm that targets arm. is it armhf or armv7l-neon? arm architecture is kinda messed up imo lol

AaronDewes, 2020-07-24:
I don't know.

AaronDewes, 2020-08-08:
I've tried building this on my RPi, while building android, my sd card broke because it used it too much. It also required 16GB of RAM, so I had to use swap.

lohriialo, 2020-09-18:
USB boot works as suggested above by @AaronDewes

closed 2020-09-18 by lohriialo

Road-Drum, 2020-09-18:
Does it works both for sd card and USB? If then you should consider pull
request for the fstab on android-rpi device repo as well

lohriialo, 2020-09-18:
No, it doesn't work both ways, I guess it has to be either SdCard or USB