android-rpi/device_arpi_rpi4 #30: Bluetooth support anytime soon?

lohriialo, 2020-08-03:
Bluetooth UI isn't available in settings of this build even though it appears there's bluetooth support in the device tree. 

I've tried this bluetooth branch https://github.com/peyo-hd/device_arpi_rpi4/tree/bt_0719 and the UI shows up in settings but it doesn't show available bluetooth devices, keeps scanning forever

Has anyone got any luck getting bluetooth working?

prashantsinhpvips, 2020-08-14:
@lohriialo 
In bt_0719 branch apply below patch for temp fix.

cd hardware/interfaces/

```
diff --git a/bluetooth/1.0/default/vendor_interface.cc b/bluetooth/1.0/default/vendor_interface.cc
index d56e344e7..51f9c8d5e 100644
--- a/bluetooth/1.0/default/vendor_interface.cc
+++ b/bluetooth/1.0/default/vendor_interface.cc
@@ -342,7 +342,7 @@ void VendorInterface::OnFirmwareConfigured(uint8_t result) {
   lib_interface_->op(BT_VND_OP_GET_LPM_IDLE_TIMEOUT, &lpm_timeout_ms);
   ALOGI("%s: lpm_timeout_ms %d", __func__, lpm_timeout_ms);
 
-  bt_vendor_lpm_mode_t mode = BT_VND_LPM_ENABLE;
+  bt_vendor_lpm_mode_t mode = BT_VND_LPM_DISABLE;
   lib_interface_->op(BT_VND_OP_LPM_SET_MODE, &mode);
 
   ALOGD("%s Calling StartLowPowerWatchdog()", __func__);

```

It worked for me.

lohriialo, 2020-08-18:
@peyo-hd thanks for the changes, latest commit now supports bluetooth, tested and working!

closed 2020-08-18 by lohriialo