android-rpi/device_arpi_rpi4 #35: Support for Tablet mode

lohriialo, 2020-09-10:
@peyo-hd is there any plan to add support for tablet mode device tree branch? TV mode has some touch/input/navigations issues when trying to use in real world application on a touch screen when there's no physical keyboard

A tablet mode branch for either arpi-10 or arpi-11 would be a really good addition

Road-Drum, 2020-09-10:
you can change the following to tablet counterpart

```
$(call inherit-product, device/google/atv/products/atv_base.mk)
...
PRODUCT_AAPT_PREF_CONFIG := tvdpi
PRODUCT_CHARACTERISTICS := tv
```

refer to older @KonstaT source code of lineage-rpi

it's probably just me but idk why `TARGET_SUPPORTS_32_BIT_APPS := false`. hopefully @peyo-hd can explain it better since android r came with native bridge support that translates ARM instruction

lohriialo, 2020-09-10:
I've tried @hkoosha tablet mode https://github.com/hkoosha/device_arpi_rpi4 but it doesn't boot and this branch is many commits behind `android-rpi`

peyo-hd, 2020-09-10:
Currently, no plan of tablet profile on this repo.
Someone could maintain it on forked branch.

Road-Drum, 2020-09-11:
@peyo-hd that's okay. anyway can you give an explanation why 32 bit app support is disabled given that there are quite a lot of them and arm64 devices doesn't have problem running arm apps?

peyo-hd, 2020-09-11:
arm 32bit enable with following commit :
https://github.com/android-rpi/device_arpi_rpi4/commit/b77892dba66ebbe0f6814427a393d344cf1c314e

lohriialo, 2020-09-16:
Tablet mode https://github.com/lohriialo/device_arpi_rpi4/tree/arpi-10-tablet

Settings UI back button doesn't work, not sure why...

lohriialo, 2020-09-18:
re-synced code base and rebuilt, back button is now working

closed 2020-09-18 by lohriialo

andersling, 2020-09-29:
@lohriialo HI,Sir! Do you have plan to move this andorid-rpi 11 to Tablet mode? Thanks for your greate job on andorid-rpi 10 tablet mode!~