android-rpi/device_arpi_rpi4 #36: App installation

csuga1, 2020-09-10:
This might be an elementary question but I had to ask. 

I followed the instructions and got the android build running on my raspberry pi 4.
Next thing I want to do is install apps... I was going to install .apk files under the file system but android file transfer (on my mac) does not recognize the android on pi (I connected pi and mac by usb, using one of the open USB ports on pi.) I set the USB configuration to MTP from the Setting, like I would normally do with my android phone, but I figured that this USB setting is for the USB-C port that's been primarily used to power pi. 

That said, what would be the best strategy to install apps on android on pi? Any help appreciated, thanks so much.

peyo-hd, 2020-09-11:
https://github.com/android-rpi/device_arpi_rpi4/wiki

csuga1, 2020-09-11:
Cool thanks!

closed 2020-09-13 by peyo-hd