android-rpi/device_arpi_rpi4 #39: Device is getting reboot  if WiFi is on

prashantsinhpvips, 2020-09-19:
Facing reboot issue when WiFi is on with **arpi-10** branch. Attaching two boots logcat logs for the reference both of the time it is rebooting at the same spot.

Note: Both files contain logs from boot to reboot.

First Boot: [reboot_logcat.txt](https://github.com/android-rpi/device_arpi_rpi4/files/5249595/reboot_logcat.txt)
Second Boot: [reboot_logcat_1.txt](https://github.com/android-rpi/device_arpi_rpi4/files/5249596/reboot_logcat_1.txt)

prashantsinhpvips, 2020-09-19:
@peyo-hd Is this issue known?

peyo-hd, 2020-09-20:
I could not find clue on the attached log.
Maybe analyze more by yourself.  Or you can compare with arpi-11 status.

prashantsinhpvips, 2020-09-21:
Thank you @peyo-hd for the valuable suggestion I have started to build arpi-11, will compare status with it.
Attaching the full logs with arpi-10 branch if it can help.
[dmesg.log](https://github.com/android-rpi/device_arpi_rpi4/files/5253595/dmesg.log)
[logcat.txt](https://github.com/android-rpi/device_arpi_rpi4/files/5253596/logcat.txt)

KonstaT, 2020-09-23:
This looks like the same issue that's been reported on my LineageOS builds (https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/6) and it's probably been present in all builds after switching to 4.19 kernel and newer (Pi 3 & Pi 4 on Android Pie & Android 10). I still have no clue why this is only reproducible on some devices/circumstances because I've never come across this myself.

prashantsinhpvips, 2020-09-23:
@KonstaT I have tried the same mage with multiple RPI4 (4GB and 8GB RAM) both are crashing. Are

 there any chances that build/build-machine has some issues? This the link of images I have used, if you want to reproduce it to your end - [link](https://drive.google.com/file/d/1FcM4L9bEd-NmUbxUjf0E_LB7FOeaGNrM/view?usp=sharing)

KonstaT, 2020-09-23:
I've never been able to reproduce the issue on any of my devices or builds. No, it's not about the build setup because the images I've released work fine for me but some small percentage of users have reported this issue using the very same images.

More likely something to do with your network environment. Wifi region, router settings, what other wifi devices are present (crash has something to do with p2p?), etc.

prashantsinhpvips, 2020-09-23:
My network has only 2 devices lapton and RPi4, Let me try it with different wifi channels.

KonstaT, 2020-09-24:
Yeah, that's also what I've been told. Maybe radio environment would better describe what I meant i.e. all wifi device/networks that are visible to your device. But there should be some common factor why this only happens on certain circumstances.

andersling, 2020-09-28:
> Yeah, that's also what I've been told. Maybe radio environment would better describe what I meant i.e. all wifi device/networks that are visible to your device. But there should be some common factor why this only happens on certain circumstances.

Hi, MR.KonstaT, thanks for your amazing work of lineage-rpi ! Do you  have plan when to update it the android 11? If you can publish all the source code (especially the  HOWTO build instructions),That is too much great!!!

and btw, The wifi problem maybe is the drivers's problem.  My device works very nice on this room~

thanks KonstaT!

@KonstaT

KonstaT, 2020-09-28:
@andersling 
Sorry, but android-rpi issue tracker is certainly not the right place to discuss my plans on lineage-rpi.

Yes, it's most definitely a driver issue. Works great on what ROM? This issue is about it not working on android-rpi which seems to be exactly the same issue as reported on my LineageOS builds.

andersling, 2020-09-29:
@KonstaT Thanks for your reply!
  I means that this andorid-rpi 11 room and the android-rpi 10( maybe this android 10 room's wifi driver was fixed by @peyo-hd  and now boots up ok with wifi ON ,on my raspberry PI4B 2G/4G/8G recently).
AS for this andorid-rpi 11 room , It runs almost pefectly on my raspberry PI4B 2G/4G/8G ,with WIFI and bluetooth ON at boot!  And the bluetooth auto connect to the Bluetooth Device(such as bluetooth audio speaker)that It connected the last time .
   I hope this information may be referenced by you to solve lineage-rpi's  wifi problem. I Like your room so much, IT'S REALLY NICE !  
   Have a good day, MR. @KonstaT:-)

KonstaT, 2020-09-29:
@andersling
Are you sure that you can reproduce this with any build anymore? You also said it's fixed with latest kernel from lineage-rpi (https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/5#issuecomment-686466563). I highly doubt that because people have still reported it with the very same kernel.

There hasn't been any recent changes to arpi-10 branch. This issue report is about latest arpi-10 branch having this issue (while android-10.0.0_r39 maybe didn't).

@peyo-hd
Any idea about this issue?

Instead of force pushing kernel branches would it be possible to keep the old source in a separate branch? It's now impossible to say e.g. what kernel source was used when manifest was at android-10.0.0_r39. This would greatly help bisecting the source code in situations like this if an issue is reproducible in one branch or an update has caused a regression. I have also enabled LOCALVERSION_AUTO on my kernels so anyone can easily see where the git HEAD was when the kernel was built.

peyo-hd, 2020-09-29:
Hi Konsta,

Kernel git in android-rpi is just for holding a few patches upon latest rpi & aosp kernel source.
Your suggestion is reasonable for storing history of release.
But I will keep it minimal in android-rpi.

andersling, 2020-09-30:
@KonstaT 
Hi,MR.KonstaT, This morning  I try to rebuild the android-rpi 10 with andorid 10 R41, it's very confusing that It can not boot up by default config !  It keeping rebooting after the rainbow screen!
and I try the Omini Room  from https://github.com/maxwen/android_device_brcm_rpi4
It compile it from source, And it boot upnormally and can connect to WIFI . It's image is arm64 bit ,so is this android-rpi 11 by MR. peyo-hd.    Is this the key reason for the 32 bit wifi failed?  @KonstaT @peyo-hd 

   BTW, I will  try to reproduce the log for you . 

    Thanks  MR.Konsta and MR. peyo-hd !

KonstaT, 2020-09-30:
@andersling No, nothing to do with architecture.

NumberOneGit, 2021-04-12:
The common aspect of most of these errors is the presence of an ANQP / Hotspot 2.0 network in local results.  This was noted by KonstaT in a LineageOS issue thread on the same topic.  I'm just updating this thread with the same info in case developers may have any possible workaround.  I've tried multiple versions of the RPi4 and CM4 with WiFi and had the same result.  I've tried in two different locations in NYC where these types of networks are prevalent and get an immediate reboot with Android-rpi and LineageOS and an incredibly unstable OmniRom which occasionally allows me to click a few menus before also rebooting.  All work fine over Ethernet with WiFi disabled.
Is there a way to disable network scan but retain access to our own WiFi network using a wpa_supplicant.conf?  The scan itself seems to be the point at which this bug occurs.

matthuo333, 2021-04-26:
In the network detection link in the initialization process, can I add a manual method of adding Ethernet IP?

kaushal98, 2022-05-19:
We have a private fork of the `arpi-11` branch and seeing this issue. We don't know the exact cause, but we also had a hotspot AP and as soon as the device rebooted it would crash within 20-30 seconds and something would keep on rebooting for 20-30 times in a row. As a temporary workaround we ended up re-flashing the SD card with new image, and not touch the Wifi settings at all and only connect using ethernet.

If it helps someone debug, this is what the log looked like when ethernet was connected and we tried to enable wifi from settings and crash/reboot.

```
05-19 02:23:47.982  2294  2294 I CarUxRManager: Context returns display ID 0
05-19 02:23:48.084  2294  2317 W OpenGLRenderer: Failed to choose config with EGL_SWAP_BEHAVIOR_PRESERVED, retrying without...
05-19 02:23:48.123  2294  2294 W View    : requestLayout() improperly called by androidx.constraintlayout.widget.ConstraintLayout{4819671 V.E...... ......ID 0,0-98,555 #7f0a009f app:id/car_ui_scroll_bar} during layout: running second layout pass
05-19 02:23:48.124  2294  2294 W View    : requestLayout() improperly called by android.view.View{369ac56 V.ED..... ......ID 46,80-52,635 #7f0a00a2 app:id/car_ui_scrollbar_thumb} during layout: running second layout pass
05-19 02:23:48.124  2294  2294 W View    : requestLayout() improperly called by com.android.car.ui.recyclerview.CarUiRecyclerView{704c4d7 V.ED..... ......ID 0,0-1084,555 #7f0a01c4 app:id/recycler_view} during layout: running second layout pass
05-19 02:23:48.140   175   175 I hwservicemanager: getTransport: Cannot find entry android.hardware.graphics.mapper@4.0::IMapper/default in either framework or device manifest.
05-19 02:23:48.140  2294  2317 I Gralloc4: mapper 4.x is not supported
05-19 02:23:48.141   175   175 I hwservicemanager: getTransport: Cannot find entry android.hardware.graphics.mapper@3.0::IMapper/default in either framework or device manifest.
05-19 02:23:48.141  2294  2317 W Gralloc3: mapper 3.x is not supported
05-19 02:23:48.262   242   425 D PermissionCache: checking android.permission.READ_FRAME_BUFFER for uid=1000 => granted (316 us)
05-19 02:23:48.294  2294  2294 W View    : requestLayout() improperly called by android.view.View{369ac56 V.ED..... ........ 46,164-52,289 #7f0a00a2 app:id/car_ui_scrollbar_thumb} during second layout pass: posting in next frame
05-19 02:23:48.302   175   175 I hwservicemanager: getTransport: Cannot find entry android.hardware.graphics.mapper@4.0::IMapper/default in either framework or device manifest.
05-19 02:23:48.302   555   582 I Gralloc4: mapper 4.x is not supported
05-19 02:23:48.303   175   175 I hwservicemanager: getTransport: Cannot find entry android.hardware.graphics.mapper@3.0::IMapper/default in either framework or device manifest.
05-19 02:23:48.304   555   582 W Gralloc3: mapper 3.x is not supported
05-19 02:23:48.323   387   412 I ActivityTaskManager: Displayed com.android.car.settings/.Settings_Launcher_Homepage: +1s42ms
05-19 02:23:48.870   555   582 I CAR.AM  : onForegroundActivitiesChanged uid 1010061 pid 1364 fg false
05-19 02:23:48.875   555   776 I CAR.AM  : Updating top task to: TaskInfoContainer [topActivity=ComponentInfo{com.android.car.mapsplaceholder/com.android.car.mapsplaceholder.MapsPlaceholderActivity}, taskId=1000018, stackId=1000018, userId=10, displayId=1, position=0
05-19 02:23:48.875   555   776 I CAR.AM  : Updating top task to: TaskInfoContainer [topActivity=ComponentInfo{com.android.car.settings/com.android.car.settings.Settings_Launcher_Homepage}, taskId=1000020, stackId=1000020, userId=10, displayId=0, position=2
05-19 02:23:48.876   555   776 I CAR.AM  : Notifying about top task: TaskInfoContainer [topActivity=ComponentInfo{com.android.car.settings/com.android.car.settings.Settings_Launcher_Homepage}, taskId=1000020, stackId=1000020, userId=10, displayId=0, position=2
05-19 02:23:48.877   555   776 I CAR.AM  : Notifying about top task: TaskInfoContainer [topActivity=ComponentInfo{com.android.car.mapsplaceholder/com.android.car.mapsplaceholder.MapsPlaceholderActivity}, taskId=1000018, stackId=1000018, userId=10, displayId=1, position=0
05-19 02:23:48.938   387  2322 W OpenGLRenderer: Failed to choose config with EGL_SWAP_BEHAVIOR_PRESERVED, retrying without...
05-19 02:23:48.000   387   406 I chatty  : uid=1000(system) android.io identical 1 line
05-19 02:23:48.012   387   406 I DropBoxManagerService: add tag=system_app_strictmode isTagEnabled=true flags=0x2
05-19 02:23:50.453   387   401 I ActivityTaskManager: START u10 {cmp=com.android.car.settings/.common.CarSettingActivities$NetworkAndInternetActivity} from uid 1001000
05-19 02:23:50.477   387   401 W ActivityTaskManager: Tried to set launchTime (0) < mLastActivityLaunchTime (132330)
05-19 02:23:50.555  2294  2294 I TetheringManager: registerTetheringEventCallback:com.android.car.settings
05-19 02:23:50.560  2294  2294 I TetheringManager: registerTetheringEventCallback:com.android.car.settings
05-19 02:23:50.591   555   582 I CAR.AM  : onTaskStackChanged
05-19 02:23:50.595   555   776 I CAR.AM  : Updating top task to: TaskInfoContainer [topActivity=ComponentInfo{com.android.car.mapsplaceholder/com.android.car.mapsplaceholder.MapsPlaceholderActivity}, taskId=1000018, stackId=1000018, userId=10, displayId=1, position=0
05-19 02:23:50.596   555   776 I CAR.AM  : Updating top task to: TaskInfoContainer [topActivity=ComponentInfo{com.android.car.settings/com.android.car.settings.common.CarSettingActivities$NetworkAndInternetActivity}, task [... cut ...]
01-01 00:02:27.447     0     0 F         : SET = 0, FnV = 0
01-01 00:02:27.451     0     0 F         : EA = 0, S1PTW = 0
01-01 00:02:27.454     0     0 F         : Data abort info:
01-01 00:02:27.457     0     0 F         : ISV = 0, ISS = 0x00000005
01-01 00:02:27.461     0     0 F         : CM = 0, WnR = 0
01-01 00:02:27.464     0     0 F user pgtable: 4k pages, 39-bit VAs, pgdp=000000014937c000
01-01 00:02:27.470     0     0 F         : [0000000000001218] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
```

greenozon, 2022-11-29:
Could anyonehint me how to collect required logs?
I've just hit this very strange and annoying issue on my Android phone ...
it never happened before, but right now it is reproducible 100% of the time, must be some fix/update from Lineage update servers...?