android-rpi/device_arpi_rpi4 #40: android 10 boot fail

mayingbin, 2020-09-21:
Hi，Peter Yoon：
   I find a problem when android10 boot on rpi 4b. 
   some init sub process are killed after kernel, (init: Untracked pid  ***) 
   I think，the fllow log is the key：
   v3d fec00000.v3d: MMU error from client ? (13) at 0x539b000, pte invalid. 
   Do you have some idea? thank you for your reply

[   18.347438] bcmgenet: Skipping UMAC reset
[   18.356323] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[   18.364869] bcmgenet fd580000.ethernet eth0: Link is Down
[   18.488681] type=1400 audit(1600511861.749:21): avc: denied { watch } for comm="system_server" path="/data/cache" dev="mmcblk0p4" ino=32770 scontext=u:r:system_server:s0 tcontext=u:object_r:cache_file:s0 tclass=dir permissive=1
[   18.513001] type=1400 audit(1600511862.199:22): avc: denied { watch } for comm=5573625365727669636520686F7374 path="/dev/bus/usb" dev="tmpfs" ino=528 scontext=u:r:system_server:s0 tcontext=u:object_r:usb_device:s0 tclass=dir permissive=1
[   18.756085] binder: unexpected work type, 4, not freed
[   18.769159] binder: undelivered TRANSACTION_COMPLETE
[   18.777003] binder: undelivered transaction 6126, process died.
[   20.306622] init: Service 'bootanim' (pid 321) exited with status 0
[   20.329415] v3d fec00000.v3d: MMU error from client ? (13) at 0x539b000, pte invalid
[   20.413246] init: processing action (sys.boot_completed=1) from (/init.rc:803)
[   20.425282] init: starting service 'exec 12 (/bin/rm -rf /data/per_boot)'...
[   20.440111] init: SVC_EXEC service 'exec 12 (/bin/rm -rf /data/per_boot)' pid 754 (uid 1000 gid 1000+0 context default) started; waiting...
[   20.448777] sdcardfs version 2.0
[   20.461182] sdcardfs: dev_name -> /data/media
[   20.466693] sdcardfs: gid=1015,mask=6
[   20.471587] sdcardfs: mounted on top of /data/media type ext4
[   20.490190] init: Service 'exec 12 (/bin/rm -rf /data/per_boot)' (pid 754) exited with status 0 waiting took 0.050000 seconds
[   20.504334] init: processing action (sys.boot_completed=1 && sys.logbootcomplete=1) from (/system/etc/init/bootstat.rc:78)
[   20.516536] init: starting service 'exec 13 (/system/bin/bootstat --record_boot_complete --record_boot_reason --record_time_since_factory_reset -l)'...
[   20.570059] init: Command 'exec_background - system log -- /system/bin/bootstat --record_boot_complete --record_boot_reason --record_time_since_factory_reset -l' action=sys.boot_completed=1 && sys.logbootcomplete=1 (/system/etc/init/bootstat.rc:83) took 53ms and succeeded
[   20.595313] init: processing action (sys.boot_completed=1) from (/system/etc/init/flags_health_check.rc:7)
[   22.843250] type=1400 audit(1600511863.379:27): avc: denied { getattr } for comm="HwBinder:204_2" path="/system/etc/bluetooth/bt_vendor.conf" dev="mmcblk0p2" ino=717 scontext=u:r:hal_bluetooth_default:s0 tcontext=u:object_r:system_file:s0 tclass=file permissive=1
[   22.849100] audit: audit_lost=1 audit_rate_limit=5 audit_backlog_limit=64
[   22.866737] type=1400 audit(1600511866.559:28): avc: denied { open } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   22.873330] audit: rate limit exceeded
[   22.901286] type=1400 audit(1600511866.559:28): avc: denied { open } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   22.925427] type=1400 audit(1600511866.559:29): avc: denied { getattr } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   22.949877] type=1400 audit(1600511866.559:29): avc: denied { getattr } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   22.974249] type=1400 audit(1600511866.559:30): avc: denied { map } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   22.998392] type=1400 audit(1600511866.559:30): avc: denied { map } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   23.022424] type=1400 audit(1600511866.559:31): avc: denied { read } for comm="crash_dump32" name="bluetooth_db-shm" dev="mmcblk0p4" ino=685 scontext=u:r:crash_dump:s0 tcontext=u:object_r:bluetooth_data_file:s0 tclass=file permissive=1
[   23.043410] type=1400 audit(1600511866.559:31): avc: denied { read } for comm="crash_dump32" name="bluetooth_db-shm" dev="mmcblk0p4" ino=685 scontext=u:r:crash_dump:s0 tcontext=u:object_r:bluetooth_data_file:s0 tclass=file permissive=1
[   23.064310] type=1400 audit(1600511866.559:32): avc: denied { open } for comm="crash_dump32" path="/data/user_de/0/com.android.bluetooth/databases/bluetooth_db-shm" dev="mmcblk0p4" ino=685 scontext=u:r:crash_dump:s0 tcontext=u:object_r:bluetooth_data_file:s0 tclass=file permissive=1
[   23.432422] selinux: SELinux: Skipping restorecon on directory(/data/system_ce/0)
[   23.432422]
[   23.443808] init: Async property child exited with status 0
[   23.453629] selinux: SELinux: Skipping restorecon on directory(/data/vendor_ce/0)
[   23.453629]
[   23.465117] init: Async property child exited with status 0
[   23.474917] selinux: SELinux: Skipping restorecon on directory(/data/misc_ce/0)
[   23.474917]
[   23.486408] init: Async property child exited with status 0
[   24.214016] init: Untracked pid 792 exited with status 0
[   24.231140] init: Untracked pid 794 exited with status 0
'▒
[   24.320088] apexd: Can't open /product/apex for reading : No such file or directory
[   24.557866] init: Received control message 'start' for 'gsid' from pid: 344 (system_server)
[   24.569562] init: starting service 'gsid'...
[   25.066690] audit: audit_lost=9 audit_rate_limit=5 audit_backlog_limit=64
[   25.073552] audit: rate limit exceeded
[   28.210168] type=1400 audit(1600511868.779:49): avc: denied { open } for comm="HwBinder:204_2" path="/system/etc/bluetooth/bt_vendor.conf" dev="mmcblk0p2" ino=717 scontext=u:r:hal_bluetooth_default:s0 tcontext=u:object_r:system_file:s0 tclass=file permissive=1
[   28.212525] audit: audit_lost=10 audit_rate_limit=5 audit_backlog_limit=64
[   28.233284] type=1400 audit(1600511871.929:51): avc: denied { open } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   28.233387] type=1400 audit(1600511871.929:51): avc: denied { open } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   28.233407] type=1400 audit(1600511871.929:52): avc: denied { getattr } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   28.233472] type=1400 audit(1600511871.929:52): avc: denied { getattr } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   28.240300] audit: rate limit exceeded
[   28.340886] type=1400 audit(1600511871.929:53): avc: denied { map } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[   28.365008] type=1400 audit(1600511871.929:53): avc: denied { map } for comm="crash_dump32" path="/dev/__properties__/u:object_r:hwservicemanager_prop:s0" dev="tmpfs" ino=455 scontext=u:r:crash_dump:s0 tcontext=u:object_r:hwservicemanager_prop:s0 tclass=file permissive=1
[... cut ...]
[   28.430980] type=1400 audit(1600511871.929:55): avc: denied { open } for comm="crash_dump32" path="/data/user_de/0/com.android.bluetooth/databases/bluetooth_db-shm" dev="mmcblk0p4" ino=685 scontext=u:r:crash_dump:s0 tcontext=u:object_r:bluetooth_data_file:s0 tclass=file permissive=1
[   29.458342] init: Untracked pid 1125 exited with status 0
[   29.464675] init: Untracked pid 1127 exited with status 0
'▒
[   34.334297] init: Untracked pid 1183 exited with status 0
[   34.350845] init: Untracked pid 1185 exited with status 0

mayingbin, 2020-09-21:
@peyo-hd , please help with this problem, waiting for your reply. thanks.

peyo-hd, 2020-09-21:
v3d MMU errors has been observed.  no fix yet from community. waiting it's stabilized by raspberry pi foundation.

Regarding bluetooth crash,  check following point.
https://github.com/android-rpi/device_arpi_rpi4/wiki/Android-10-:-patch-framework-source#disable-low-power-mode-of-bluetooth

Maybe need analysis by yourself.

mayingbin, 2020-09-22:
Hi，peyo-hd：
    I add android10 framework patch，
    Now bluetooth no crash. But blocked the gsid operation，and try twice timeout：
**type=1400 audit(1600511868.159:32): avc: denied { open } for comm="gsid" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=10617 scontext=u:r:gsid:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1**
    Do you have some idea？@peyo-hd

[   23.574679] init: Async property child exited with status 0
[   23.585234] selinux: SELinux: Skipping restorecon on directory(/data/vendor_ce/0)
[   23.585234]
[   23.597764] init: Async property child exited with status 0
[   23.608097] selinux: SELinux: Skipping restorecon on directory(/data/misc_ce/0)
[   23.608097]
[   23.620709] init: Async property child exited with status 0
[   23.974195] apexd: Can't open /product/apex for reading : No such file or directory
[   24.035131] type=1400 audit(1600511863.469:29): avc: denied { open } for comm="bluetooth@1.0-s" path="/lib/firmware/brcm/BCM4345C0.hcd" dev="mmcblk0p2" ino=34 scontext=u:r:hal_bluetooth_default:s0 tcontext=u:object_r:rootfs:s0 tclass=file permissive=1
[   24.062428] type=1400 audit(1600511867.779:30): avc: denied { watch } for comm="Thread-8" path="/data/tombstones" dev="mmcblk0p4" ino=8200 scontext=u:r:system_server:s0 tcontext=u:object_r:tombstone_data_file:s0 tclass=dir permissive=1
[   24.324613] init: Received control message 'start' for 'gsid' from pid: 344 (system_server)
[   24.336385] init: starting service 'gsid'...
[   24.411610] type=1400 audit(1600511867.779:30): avc: denied { watch } for comm="Thread-8" path="/data/tombstones" dev="mmcblk0p4" ino=8200 scontext=u:r:system_server:s0 tcontext=u:object_r:tombstone_data_file:s0 tclass=dir permissive=1
[   24.433967] type=1400 audit(1600511868.159:31): avc: denied { write } for comm="gsid" name="tasks" dev="tmpfs" ino=10617 scontext=u:r:gsid:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   24.435561] init: Received control message 'stop' for 'gsid' from pid: 344 (system_server)
[   24.452760] type=1400 audit(1600511868.159:31): avc: denied { write } for comm="gsid" name="tasks" dev="tmpfs" ino=10617 scontext=u:r:gsid:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   24.460551] init: Sending signal 9 to service 'gsid' (pid 944) process group...
**[   24.478629] type=1400 audit(1600511868.159:32): avc: denied { open } for comm="gsid" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=10617 scontext=u:r:gsid:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1
[   78.335983] type=1400 audit(1600511868.159:32): avc: denied { open } for comm="gsid" path="/dev/stune/foreground/tasks" dev="tmpfs" ino=10617 scontext=u:r:gsid:s0 tcontext=u:object_r:device:s0 tclass=file permissive=1**
[   78.355771] type=1400 audit(1600511922.079:33): avc: denied { associate } for comm="Binder:199_3" name="globalAlert" scontext=u:object_r:proc_net:s0 tcontext=u:object_r:proc:s0 tclass=filesystem permissive=1

mayingbin, 2020-09-22:
hi，@peyo-hd ：
And android11，boot process is blocked，no log after **29.931911**
It is so weird，the log don't have crash or process died，
Do you have some idea？the whole log: [log3.txt](https://github.com/android-rpi/device_arpi_rpi4/files/5261333/log3.txt)
waiting for your reply.

[   18.141753] init: processing action (sys.sysctl.extra_free_kbytes=*) from (/system/etc/init/hw/init.rc:1002)
[   20.294058] bcmgenet: Skipping UMAC reset
[   20.300257] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[   20.308838] bcmgenet fd580000.ethernet eth0: Link is Down
[   20.351035] type=1400 audit(1600768880.403:24): avc: denied { read } for comm="Binder:183_2" name="wakeup0" dev="sysfs" ino=13590 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   20.371303] type=1400 audit(1600768880.403:25): avc: denied { open } for comm="Binder:183_2" path="/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0" dev="sysfs" ino=13590 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   21.874913] type=1400 audit(1600768881.927:26): avc: denied { read } for comm="Binder:183_2" name="wakeup0" dev="sysfs" ino=13590 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   21.895255] type=1400 audit(1600768881.927:27): avc: denied { open } for comm="Binder:183_2" path="/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0" dev="sysfs" ino=13590 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   21.924969] type=1400 audit(1600768881.927:28): avc: denied { read } for comm="Binder:183_2" name="event_count" dev="sysfs" ino=13597 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[   21.949656] type=1400 audit(1600768881.927:29): avc: denied { open } for comm="Binder:183_2" path="/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0/event_count" dev="sysfs" ino=13597 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[   21.978898] type=1400 audit(1600768881.927:30): avc: denied { getattr } for comm="Binder:183_2" path="/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0/event_count" dev="sysfs" ino=13597 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[   22.028941] type=1400 audit(1600768882.083:31): avc: denied { associate } for comm="Binder:202_4" name="globalAlert" scontext=u:object_r:proc_net:s0 tcontext=u:object_r:proc:s0 tclass=filesystem permissive=1
[   22.214428] type=1400 audit(1600768882.267:32): avc: denied { read } for comm="Binder:183_2" name="wakeup0" dev="sysfs" ino=13590 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   22.237961] type=1400 audit(1600768882.267:33): avc: denied { open } for comm="Binder:183_2" path="/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0" dev="sysfs" ino=13590 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   23.288194] uart-pl011 fe201000.serial: no DMA platform data
[   23.363051] perfetto: enabled ftrace
[   23.806758] init: Service 'bootanim' (pid 333) exited with status 0 oneshot service took 15.677000 seconds in background
[   23.819201] init: Sending signal 9 to service 'bootanim' (pid 333) process group...
[   23.827861] libprocessgroup: Successfully killed process cgroup uid 1003 pid 333 in 0ms
[   23.911173] init: processing action (sys.boot_completed=1) from (/system/etc/init/hw/init.rc:993)
[   23.924505] init: starting service 'exec 14 (/bin/rm -rf /data/per_boot)'...
[   23.943365] init: SVC_EXEC service 'exec 14 (/bin/rm -rf /data/per_boot)' pid 831 (uid 1000 gid 1000+0 context default) started; waiting...
[   23.958911] v3d fec00000.v3d: MMU error from client ? (13) at 0x4a5b000, pte invalid
[   24.017501] init: Service 'exec 14 (/bin/rm -rf /data/per_boot)' (pid 831) exited with status 0 waiting took 0.074000 seconds
[   24.029413] init: Sending signal 9 to service 'exec 14 (/bin/rm -rf /data/per_boot)' (pid 831) process group...
[   24.040376] libprocessgroup: Successfully killed process cgroup uid 1000 pid 831 in 0ms
[   24.050723] init: processing action (sys.boot_completed=1 && sys.bootstat.first_boot_completed=0) from (/system/etc/init/bootstat.rc:77)
[   29.931911] perfetto: disabled ftrace

mayingbin, 2020-09-22:
I use pi 4B(8G Ram) and official 7” display (800*480) touch screen.

mayingbin, 2020-09-23:
Hi, android-rpi teams：
  Is the follow log related to the display？**[drm] Cannot find any crtc or sizes**
  when boot, the 7 touch screen, display rainbow and then always black.

mayingbin, 2020-09-23:
Hi, android-rpi teams：
    Finally，android boot successfully with hdmi，but doesn‘t support the offical 7” display (800*480) touch screen with dsi.
    （[drm] Cannot find any crtc or sizes）
    Do you encounter this problem，have some idea to support dsi display？

saurabhin25, 2020-09-23:
Change the density to 120 in build.prop

mayingbin, 2020-09-23:
@saurabhin25  which build.prop，could you give the dir？

saurabhin25, 2020-09-23:
Under vendor

mayingbin, 2020-09-23:
Hi，saurabhin25，
       vendor dir just have RpLauncher APP src.
        After android boot，settingprovider just don’t have the smallest width options.

the setting：
[image]

mayingbin, 2020-09-23:
For 7” display (800*480) touch screen， I change device/arpi/rpi4/rpi4.mk debug.drm.mode.force 1280x720 by default to 800x480,
when android boot，still **[drm] Cannot find any crtc or sizes**
whether I should config something other？

saurabhin25, 2020-09-23:
Install super browser and find build.prop
Edit that file change density to 120

Or use any linux pc insert card and find the build.prop file under vendor
edit that using notepad++

KonstaT, 2020-09-23:
KMS driver doesn't support DSI connector the official display panel uses. You would need to ask Pi Foundation what their plans are with this.

mayingbin, 2020-09-23:
enn，thanks for your reply.

sebastianene07, 2020-12-06:
> KMS driver doesn't support DSI connector the official display panel uses. You would need to ask Pi Foundation what their plans are with this.

I am also interested in this. Maybe we should try and port this from the Raspbian image

KonstaT, 2020-12-06:
> > KMS driver doesn't support DSI connector the official display panel uses. You would need to ask Pi Foundation what their plans are with this.
> 
> I am also interested in this. Maybe we should try and port this from the Raspbian image

Raspberry Pi OS uses FKMS (that support DSI), not KMS that is used on Android.

There's was recently some patches merged for this on 5.9/5.10 kernel (https://github.com/raspberrypi/linux/pull/3954). You can try to backport to 5.4 if you like.

sebastianene07, 2020-12-06:
> > > KMS driver doesn't support DSI connector the official display panel uses. You would need to ask Pi Foundation what their plans are with this.
> > 
> > 
> > I am also interested in this. Maybe we should try and port this from the Raspbian image
> 
> Raspberry Pi OS uses FKMS (that supports DSI), not KMS that is used on Android.
> 
> There's was recently some patches merged for this on 5.9/5.10 kernel ([raspberrypi/linux#3954](https://github.com/raspberrypi/linux/pull/3954)). You can try to backport to 5.4 if you like.

Magic, thank you !