android-rpi/device_arpi_rpi4 #41: Is it possible to output audio with HDMI?

jaeminkang2, 2020-09-22:
Hello team

I have a question.. 
Last time someone else asked the same question, 
(https://github.com/android-rpi/device_arpi_rpi4/issues/19#issuecomment-671923796)

Can the rpi4 output audio from hdmi?

If you have a guide or hint, could you share it ?

peyo-hd, 2020-09-22:
Refer to this : https://github.com/maxwen/android_hardware_arpi/commit/76add18198b2a7693a54e2f4e30bc20281b6aa8c

jaeminkang2, 2020-09-22:
Hi peyo, 
Thanks for your guide.  I will check refer to your share site.

KonstaT, 2020-09-23:
Since maxwen has replicated what I've done on my LineageOS builds, I'll comment on this. https://github.com/maxwen/proprietary_vendor_brcm/commit/269d6e582cbcb42939054f9e4b75ec9855128896 is much more relevant to this. You can either add the 'noaudio' parameter to VC4 or revert https://github.com/android-rpi/kernel_arpi/commit/48fda0f0ffdb5bbfa22f61514bd65235064a0f57 for the bcm2835_audio HDMI interface to be available. That should be enough with the audio HAL that's used on android-rpi.

https://github.com/maxwen/android_hardware_arpi/commit/76add18198b2a7693a54e2f4e30bc20281b6aa8c is useful if you want to switch between HDMI and 3.5 jack or even use an external audio DACs. How this is handled was changed with the 5.4 kernel.

jaeminkang2, 2020-09-24:
Hi KonstaT
I have been checking maxwen & android-rpi git source since yesterday.
Your guide will be very useful.
Thanks for your guidance.

closed 2020-09-27 by jaeminkang2