android-rpi/device_arpi_rpi4 #43: Thanks for Peter Yoon's greate job! Some isues to report to u:)

andersling, 2020-09-28:
Hi,MR. Peter Yoon! Thanks for your greate efforct& up to date android 11 nice work for pi!
I build it follow by your instructions ,it takes about 5 hours. and I burn it to RPI4b, it's so exciting that it can boot up normally,even with wifi and bluetooth are ON! and WIFI can connect to AP(ac or abgn) ,and the bluetooth is also ok when connect to some sound speakers.
   Most import of all, The user experience (such as the mouse move quickly,or desktop screen update etc) is very excellent on this ARM64 version!~
    but after a day's test,i found some isues:
    A. when play some 1920x1080p videos (MP4),the sound is OK,but the screen keeping frozen for some seconds and then jump to the cureent frame,and after a few seconds, it frozen again and again!
    B. How to root the device by default? 
    C. Can I change the screen's resolution to 1920x1080? How to do it?Thanks

   Thanks for all the 3 persons of the project!  Your greate job make it amazing!~!!!!

   @peyo-hd

peyo-hd, 2020-09-29:
Hi andersling,

For 1080p, you can change debug.drm.mode.force in rpi.mk.
Select 720p for video, It seems AOSP SW decoder has issues.

android-rpi build is mostly un-touched AOSP,
so you can look into the source or get help from web.

andersling, 2020-09-30:
Thanks for MR. peyo-hd 's  reply!!
   flowing your instruction, I change debug.drm.mode.force in rpi.mk from 1280x720 to 1920x1080, and rebuid the room, but It boot failed (after the rainbow screen of RaspberryPI,it became black for 2-3 seconds, and the LED monitor show that NO SINGAL, and RPI4 go to the rainbow screen again and again...
   
   and as for the root user access isue, I found a solution from  stackoverflow:
      https://stackoverflow.com/questions/5597139/build-android-with-superuser
  but rebuild the image and try to boot from the new room,it seems that the terminal still  can't to switch to su (It said that "permittion denied")

  any idea?  Thanks~

@peyo-hd