android-rpi/device_arpi_rpi4 #45: Deprecated usage of envsetup.sh

mashua, 2020-10-12:
Tried to build an image for a rapi 4 following your instructions, but at the step where we need to source the envsetup.sh 
the file emits the following warning:

echo "Sourcing 'build/envsetup.sh' for Android Kernels is deprecated and no longer valid!"
echo "Please refer to the documentation in said script for details."

The referred documentation page is the:
https://source.android.com/setup/build/building-kernels
which contains not so easy to follow instructions.

Any thoughs / updates on this ?

schneideris, 2020-10-12:
Try running the following:

`repo sync --force-sync`

mashua, 2020-10-14:
Thanks for the reply @schneideris , but this didn;t help.
The main reason here is that the upstream AOSP branch named android-11.0.0_r5 has changed the way that it builds its image and the instructions are no longer relevant and up to date. 

Do anyone has an indication of the latest AOSP upstream branch that the instructions work ?

mashua, 2020-10-14:
So it seems that currently a lot is going re: AOSP build system. 
Its a hybrid situation where make , envsetup.sh, lunch and soong co-exists and this creates some confusion for newcomers like me.
The repo instructions are OK, thus closing

closed 2020-10-14 by mashua