android-rpi/device_arpi_rpi4 #46: camera support

mayingbin, 2020-10-14:
Hi，android-rpi team：
     I use RpCamera APP, but the app crash when app boot. Whether the code support camera？
     The crash info as follows， please have a look the log txt. 

6904 10-14 07:21:43.850   226   816 E CameraService: cameraIdIntToStrLocked: input id 0 invalid: valid range  (0, 0)
6905 10-14 07:21:43.850   226   816 I CameraService: CameraService::connect call (PID -1 "com.arpi.rpcamera", camera ID ) for HAL version default and Camera API version 1
6906 10-14 07:21:43.850   226   816 W CameraService: Attempting to connect to system-only camera id , connection rejected
6907 10-14 07:21:43.851  1622  1622 W CameraBase: An error occurred while connecting to camera 0: Status(-8, EX_SERVICE_SPECIFIC): '4: connectHelper:1418: No camera device with ID "" currently available'
6908 10-14 07:21:43.852  1622  1622 D AndroidRuntime: Shutting down VM
6909 --------- beginning of crash
6910 10-14 07:21:43.854  1622  1622 E AndroidRuntime: FATAL EXCEPTION: main
6911 10-14 07:21:43.854  1622  1622 E AndroidRuntime: Process: com.arpi.rpcamera, PID: 1622
6912 10-14 07:21:43.854  1622  1622 E AndroidRuntime: java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.arpi.rpcamera/com.arpi.rpcamera.MainActivity}: java.lang.RuntimeException: Fail to connect to camera service
6913 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.ActivityThread.deliverResults(ActivityThread.java:4845)
6914 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.ActivityThread.handleSendResult(ActivityThread.java:4886)
6915 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:51)
6916 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
6917 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
6918 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
6919 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.os.Handler.dispatchMessage(Handler.java:107)
6920 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.os.Looper.loop(Looper.java:214)
6921 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.ActivityThread.main(ActivityThread.java:7356)
6922 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at java.lang.reflect.Method.invoke(Native Method)
6923 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
6924 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
6925 10-14 07:21:43.854  1622  1622 E AndroidRuntime: Caused by: java.lang.RuntimeException: Fail to connect to camera service
6926 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.hardware.Camera.<init>(Camera.java:551)
6927 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.hardware.Camera.open(Camera.java:396)
6928 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at com.arpi.rpcamera.MainActivity.openCamera(MainActivity.kt:58)
6929 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at com.arpi.rpcamera.MainActivity.onRequestPermissionsResult(MainActivity.kt:49)
6930 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:8289)
6931 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.Activity.dispatchActivityResult(Activity.java:8139)
6932 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     at android.app.ActivityThread.deliverResults(ActivityThread.java:4838)
6933 10-14 07:21:43.854  1622  1622 E AndroidRuntime:     ... 11 more
6934 10-14 07:21:43.940   341  1656 I DropBoxManagerService: add tag=system_app_crash isTagEnabled=true flags=0x2
6935 10-14 07:21:43.940   341   981 W ActivityTaskManager:   Force finishing activity com.arpi.rpcamera/.MainActivity
6936 10-14 07:21:43.952  1622  1622 I Process : Sending signal. PID: 1622 SIG: 9
6937 10-14 07:21:43.986   341   416 W InputDispatcher: channel '5b95b24 com.arpi.rpcamera/com.arpi.rpcamera.MainActivity (server)' ~ Consumer closed input channel or an error occurred.  events=0x9
6938 10-14 07:21:43.986   341   416 E InputDispatcher: channel '5b95b24 com.arpi.rpcamera/com.arpi.rpcamera.MainActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
6939 10-14 07:21:43.987   341   980 I WindowManager: WIN DEATH: Window{5b95b24 u0 com.arpi.rpcamera/com.arpi.rpcamera.MainActivity}
6940 10-14 07:21:43.988   341   980 W InputDispatcher: Attempted to unregister already unregistered input channel '5b95b24 com.arpi.rpcamera/com.arpi.rpcamera.MainActivity (server)'
6941 10-14 07:21:43.989   341   359 I ActivityManager: Process com.arpi.rpcamera (pid 1622) has died: fore TOP 
6942 10-14 07:21:43.998   197   197 I Zygote  : Process 1622 exited due to signal 9 (Killed)
6943 10-14 07:21:44.008   341   363 W ActivityManager: setHasOverlayUi called on unknown pid: 1622
6944 10-14 07:21:44.035   341   371 I libprocessgroup: Successfully killed process cgroup uid 10014 pid 1622 in 45ms
6945 10-14 07:21:44.055   775   775 D ImageWallpaper: wallpaper visibility changes to: true
6946 10-14 07:21:44.396   218   218 W SurfaceFlinger: couldn't log to binary event log: overflow.
6947 10-14 07:21:44.442   341   363 W ActivityTaskManager: Activity top resumed state loss timeout for ActivityRecord{881d72a u0 com.arpi.rpcamera/.MainActivity t-1 f}
6948 10-14 07:21:45.268   341   980 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.arpi.rpcamera/.MainActivity bnds=[205,115][378,241]} from uid 10026
6949 10-14 07:21:45.289   341   364 E system_server: Invalid ID 0x00000000.
6950 10-14 07:21:45.364   197   197 D Zygote  : Forked child process 1658
6951 10-14 07:21:45.368   341   370 W ActivityManager: Slow operation: 79ms so far, now at startProcess: returned from zygote!
6952 10-14 07:21:45.368   341   370 W ActivityManager: Slow operation: 79ms so far, now at startProcess: done updating battery stats
6953 10-14 07:21:45.368   341   370 W ActivityManager: Slow operation: 80ms so far, now at startProcess: building log message
6954 10-14 07:21:45.368   341   370 I ActivityManager: Start proc 1658:com.arpi.rpcamera/u0a14 for activity {com.arpi.rpcamera/com.arpi.rpcamera.MainActivity}
6955 10-14 07:21:45.368   341   370 W ActivityManager: Slow operation: 80ms so far, now at startProcess: starting to update pids map
6956 10-14 07:21:45.369  1658  1658 I Zygote  : seccomp disabled by setenforce 0
6957 10-14 07:21:45.369   341   370 W ActivityManager: Slow operation: 80ms so far, now at startProcess: done updating pids map
6958 10-14 07:21:45.530   226   770 E CameraService: cameraIdIntToStrLocked: input id 0 invalid: valid range  (0, 0)
6959 10-14 07:21:45.530   226   770 I CameraService: CameraService::connect call (PID -1 "com.arpi.rpcamera", camera ID ) for HAL version default and Camera API version 1
6960 10-14 07:21:45.530   226   770 W CameraService: Attempting to connect to system-only camera id , connection rejected
6961 10-14 07:21:45.531  1658  1658 W CameraBase: An error occurred while connecting to camera 0: Status(-8, EX_SERVICE_SPECIFIC): '4: connectHelper:1418: No camera device with ID "" currently available'
6962 10-14 07:21:45.532  1658  1658 D AndroidRuntime: Shutting down VM
6963 10-14 07:21:45.533  1658  1658 E AndroidRuntime: FATAL EXCEPTION: main
6964 10-14 07:21:45.533  1658  1658 E AndroidRuntime: Process: com.arpi.rpcamera, PID: 1658
6965 10-14 07:21:45.533  1658  1658 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{com.arpi.rpcamera/com.arpi.rpcamera.MainActivity}: java.lang.RuntimeException: Fail to connect to camera service
6966 10-14 07:21:45.533  1658  1658 E AndroidRuntime:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3270)
6967 10-14 07:21:45.533  1658  1658 E AndroidRuntime:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3409)
[... cut ...]
6986 10-14 07:21:45.533  1658  1658 E AndroidRuntime:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3245)
6987 10-14 07:21:45.533  1658  1658 E AndroidRuntime:     ... 11 more
6988 10-14 07:21:45.536   341  1682 I DropBoxManagerService: add tag=system_app_crash isTagEnabled=true flags=0x2
6989 10-14 07:21:45.536   341  1657 W ActivityTaskManager:   Force finishing activity com.arpi.rpcamera/.MainActivity
[camera_fail.txt](https://github.com/android-rpi/device_arpi_rpi4/files/5377647/camera_fail.txt)

mayingbin, 2020-10-14:
I use official Pi camera modules & UVC USB webcams

peyo-hd, 2020-10-14:
Only USB camera is supported by arpi build. Tested with logitech c270.

For the failure, try applying following changes in /hardware/arpi
https://github.com/android-rpi/hardware_arpi/commit/94a84ce20decd505c473923b8bde163870e18e66

In the log 'camera_fail.txt',  following log shows 0 detection. If successful it will detect 1 device.
 672 01-01 00:00:08.232   226   226 I CameraProviderManager: Camera provider external/0 ready with 0 camera devices
(A few more seconds delay could be tried if detection not successful)

Android 11 seems working better with USB camera, then you can migrate to arpi-11 build.

mayingbin, 2020-10-15:
I test usb camera and working as well,  but any suggestion to support official Pi camera modules & UVC USB webcams？

ohmex, 2020-11-12:
Hi peyo
I appreciate the efforts put into this build.
Trying to build & add the official camera hal to your project as described here https://github.com/lineage-rpi/android_local_manifest/issues/9#issuecomment-671292548
But could not get it to work, could you please help here?
have you tried using this HAL?

vinayaksoni, 2021-04-12:
@peyo-hd - How did you test with logitech c270 on Android 11? The RpCamera (compiled in Android Studio) with no other change in the local_manifest i.e. using the external camera provider hal 2.5 ?

saurabhin25, 2021-04-25:
Is this running on official 7 inch display?