android-rpi/device_arpi_rpi4 #48: media playback performance lag

lohriialo, 2020-10-21:
I've noticed that there's considerable media playback performance lag in [arpi-11](https://github.com/android-rpi/device_arpi_rpi4/tree/arpi-10) vs [arpi-10](https://github.com/android-rpi/device_arpi_rpi4/tree/arpi-10), I was able to reliably stream youtube videos with a native android webview but the same video playback does freeze up the screen altogether in arpi-11

lohriialo, 2020-10-21:
For example when I start playing a youtube video in a native webview, I see the following errors : 
```
10-21 12:32:20.305  1313  1434 D NdkImageReader: acquireImageLocked: Overriding buffer format YUV_420_888 to 0x32315659.
10-21 12:32:20.349  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:20.354  1313  1349 E FirebaseInstanceId: Google Play services missing or without correct permission.
10-21 12:32:20.395  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:21.929  1313  1434 I chatty  : uid=10032(com.video.test.demo) Chrome_InProcGp identical 38 lines
10-21 12:32:21.962  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:21.973  1313  1345 E v3d_bufmgr: v3d_bo_from_cache(): wait failed: EINVAL
10-21 12:32:22.012  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:22.045  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:22.049  1313  1345 E v3d_bufmgr: v3d_bo_from_cache(): wait failed: EINVAL
10-21 12:32:22.096  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:23.012  1313  1434 I chatty  : uid=10032(com.video.test.demo) Chrome_InProcGp identical 23 lines
10-21 12:32:23.045  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:23.065   350   423 E WifiVendorHal: getWifiLinkLayerStats_1_3_Internal(l.973) failed {.code = ERROR_UNKNOWN, .description = unknown error}
10-21 12:32:23.098  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:23.730  1313  1434 I chatty  : uid=10032(com.video.test.demo) Chrome_InProcGp identical 16 lines
10-21 12:32:23.762  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:23.771   350   685 D BufferPoolAccessor2.0: evictor expired: 5, evicted: 0
10-21 12:32:23.812  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:24.013  1313  1434 I chatty  : uid=10032(com.video.test.demo) Chrome_InProcGp identical 5 lines
10-21 12:32:24.046  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:24.055  1313  1473 D BufferPoolAccessor2.0: bufferpool2 0x738863ee28 : 7(14680064 size) total buffers - 3(6291456 size) used buffers - 96/103 (recycle/alloc) - 13/204 (fetch/transfer)
10-21 12:32:24.095  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:24.728  1313  1434 I chatty  : uid=10032(com.video.test.demo) Chrome_InProcGp identical 16 lines
10-21 12:32:24.761  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:24.772   350   685 D BufferPoolAccessor2.0: evictor expired: 1, evicted: 0
10-21 12:32:24.812  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:26.012  1313  1434 I chatty  : uid=10032(com.video.test.demo) Chrome_InProcGp identical 30 lines
10-21 12:32:26.046  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:26.079   350   423 E WifiVendorHal: getWifiLinkLayerStats_1_3_Internal(l.973) failed {.code = ERROR_UNKNOWN, .description = unknown error}
10-21 12:32:26.097  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:29.012  1313  1434 I chatty  : uid=10032(com.video.test.demo) Chrome_InProcGp identical 73 lines
10-21 12:32:29.046  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:29.089   350   423 E WifiVendorHal: getWifiLinkLayerStats_1_3_Internal(l.973) failed {.code = ERROR_UNKNOWN, .description = unknown error}
10-21 12:32:29.101  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:29.131  1313  1434 E chromium: [ERROR:android_image_reader_utils.cc(19)] Failed to create android native fence sync object.
10-21 12:32:29.134   350  1307 W OpenGLRenderer: Failed to choose config with EGL_SWAP_BEHAVIOR_PRESERVED, retrying without...
```

peyo-hd, 2020-10-24:
It seems WebView video-player is not working properly with vp9 decoder allowed in arpi-11 build.
You can try disabling vp9 decoder as described in arpi-10 patch : 
https://github.com/android-rpi/device_arpi_rpi4/wiki/Android-10-:-patch-framework-source#enable-legacy-sw-video-decoder
Apply only one point of  deleting "libcodec2_soft_vp9dec.so" line.

You might have tested that Cobalt app mentioned in below link - is ok with the vp9 decoder.
https://github.com/android-rpi/device_arpi_rpi4/wiki/arpi-11-:-framework-patch#enable-cameraserver--mediaserver-64bit-build

lohriialo, 2020-10-30:
@peyo-hd thanks for the suggestion, I got a chance to try this today by removing just the single line `emplace("libcodec2_soft_vp9dec.so");` from `frameworks/av/media/codec2/vndk/C2Store.cpp` but that didn't help either, following is the logcat

```
10-30 08:23:26.637   237   303 W HidlServiceManagement: Waited one second for android.hardware.media.omx@1.0::IOmxStore/default
10-30 08:23:26.637   163   163 I hwservicemanager: Since android.hardware.media.omx@1.0::IOmxStore/default is not registered, trying to start it as a lazy HAL.
10-30 08:23:26.638   237   303 I HidlServiceManagement: getService: Trying again for android.hardware.media.omx@1.0::IOmxStore/default...
10-30 08:23:26.654   163  2030 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.media.omx@1.0::IOmxStore/default": error code: 0x20
01-01 00:06:32.433     0     0 E init    : Control message: Could not find 'android.hardware.media.omx@1.0::IOmxStore/default' for ctl.interface_start from pid: 163 (/system/bin/hwservicemanager)
01-01 00:06:33.153     0     0 I init    : starting service 'vendor.media.omx'...
10-30 08:23:27.452  2031  2031 I android.hardware.media.omx@1.0-service: mediacodecservice starting
10-30 08:23:27.452  2031  2031 W android.hardware.media.omx@1.0-service: Could not read additional policy file '/vendor/etc/seccomp_policy/mediacodec.policy'
10-30 08:23:27.453  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: failed to get path of fd 4: No such file or directory
10-30 08:23:27.454  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: allowing syscall: connect
10-30 08:23:27.454  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: allowing syscall: fcntl
10-30 08:23:27.454  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: allowing syscall: sendto
10-30 08:23:27.454  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: allowing syscall: socket
10-30 08:23:27.454  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: allowing syscall: writev
10-30 08:23:27.454  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(14): nonexistent syscall 'mmap2'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(18): nonexistent syscall 'ftruncate64'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(34): nonexistent syscall 'getuid32'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(35): nonexistent syscall 'fstat64'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(36): nonexistent syscall 'fstatfs64'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(51): nonexistent syscall '_llseek'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(54): nonexistent syscall 'statfs64'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(56): nonexistent syscall 'fstatat64'
10-30 08:23:27.455  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: <fd>(57): nonexistent syscall 'ugetrlimit'
10-30 08:23:27.456  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(5): nonexistent syscall 'sigreturn'
10-30 08:23:27.456  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(31): nonexistent syscall 'getuid32'
10-30 08:23:27.456  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(32): nonexistent syscall 'fstat64'
10-30 08:23:27.456  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(33): nonexistent syscall 'mmap2'
10-30 08:23:27.456  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(34): nonexistent syscall 'geteuid32'
10-30 08:23:27.457  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(35): nonexistent syscall 'getgid32'
10-30 08:23:27.457  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(36): nonexistent syscall 'getegid32'
10-30 08:23:27.457  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: compile_file: /system/etc/seccomp_policy/crash_dump.arm.policy(37): nonexistent syscall 'getgroups32'
10-30 08:23:27.457  2031  2031 W android.hardware.media.omx@1.0-service: libminijail[2031]: logging seccomp filter failures
10-30 08:23:27.459  2031  2031 E android.hardware.media.omx@1.0-service: libminijail[2031]: blocked syscall: mmap
01-01 00:06:33.255     0     0 I init    : Service 'vendor.media.omx' (pid 2031) received signal 31
01-01 00:06:33.263     0     0 I init    : Sending signal 9 to service 'vendor.media.omx' (pid 2031) process group...
[... cut ...]
10-30 08:23:31.663   163  2036 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.media.omx@1.0::IOmxStore/default": error code: 0x20
10-30 08:23:32.147   345   424 E WifiVendorHal: getWifiLinkLayerStats_1_3_Internal(l.973) failed {.code = ERROR_UNKNOWN, .description = unknown error}
01-01 00:06:36.439     0     0 I chatty  : uid=0(root) logd identical 3 lines
01-01 00:06:37.441     0     0 E init    : Control message: Could not find 'android.hardware.media.omx@1.0::IOmxStore/default' for ctl.interface_start from pid: 163 (/system/bin/hwservicemanager)
```

peyo-hd, 2020-10-30:
I seems you omit following step in https://github.com/android-rpi/device_arpi_rpi4/wiki/arpi-11-:-framework-patch#enable-cameraserver--mediaserver-64bit-build

     Replace following file under frameworks/av with mediacodec-arm64.policy

     services/mediacodec/seccomp_policy/mediacodec-arm64.policy

lohriialo, 2020-10-30:
I have applied that patch, I replaced frameworks/av/services/mediacodec/seccomp_policy/mediacodec-arm64.policy with the file download from that google drive

lohriialo, 2020-10-30:
Btw, Cobalt video test works great