android-rpi/device_arpi_rpi4 #51: Wifi not turning on with the new Raspberry Pi 400

lohriialo, 2020-11-09:
I just tried arpi-10 on the new Raspberry Pi 400, wifi won't turn on. Was hoping for it to work out of the box with an Rpi4 build but turns out it's not, I believe Raspberry Pi 400 has a newer chip than the Pi4

gioan777, 2020-11-09:
Coming from Arch Linux ARM with the same issue. In my case (from running `dmesg | grep mmc1` mmc1 being the wifi chip) it seems that it tries to load the file `/lib/firmware/brcm/brcmfmac43456-sdio.bin` but that file doesn't exist in the distro.

 After I copied the Raspberry Pi 400 firmware files (`/lib/firmware/brcm/brcmfmac43456-sdio.*` ) from the [latest official Raspberry Pi OS iso](https://downloads.raspberrypi.org/raspios_lite_armhf_latest) to my installation, wifi worked.

From what I see in this repo copying the firmware files to `device_arpi_rpi4/firmware/brcm/` may also do the trick.

peyo-hd, 2021-05-31:
Wifi is now working with arpi-12 build.