android-rpi/device_arpi_rpi4 #57: Stuck at booting animation

ezebongiovi, 2020-12-30:
# Description
I've built the Android OS by following the instructions on this repository. All went well but when the RPI boots it gets stuck on the android logo (booting animation)

# Notes
The system image was bigger than `1024M` so I made some changes at `device/arpi/rpi4/BoardConfig.mk` to extending each partition. The file looks like this now:

[image]

# Environment

The **SD Card partitions** look like this

[image]

The **SD Card volumes** look like this
[image]

## Device
Raspberry PI 4 (4GB)

# Question
@peyo-hd is there something I'm doing wrong?

ezebongiovi, 2020-12-31:
I was able to boot with `rpi-10` branch

shishui324, 2021-07-13:
I also ran into the same problem

cod3monk3y, 2021-09-18:
I am also experiencing this. Boot is stuck and android logo. Logo is animating light/dark, so it doesn't seem frozen. Neither adb nor fastboot show any devices at this stage. Any advice on how to diagnose this?

sumikchakraborty, 2022-03-20:
What is the RAM of the Raspberri Pi ?

sumikchakraborty, 2022-03-20:
> I was able to boot with `rpi-10` branch

What RAM was able to boot successfully ?