android-rpi/device_arpi_rpi4 #60: ADB missing device over USB

dananas, 2021-01-25:
Hi, is it possible to connect to RPi with ADB using USB?

I've tried changing USB configuration / USB debug properties in Settings and I still don't see device in **adb devices**

peyo-hd, 2021-01-26:
Refer to https://github.com/android-rpi/device_arpi_rpi4/commit/ad376c69328339c381b9d51680f4431faf9b70b6

dananas, 2021-01-30:
Thanks, it works now! For those who have the same problems:
- Make sure that the code from this two commits is present in your branch [ad376c6](https://github.com/android-rpi/device_arpi_rpi4/commit/ad376c69328339c381b9d51680f4431faf9b70b6), [8b86126](https://github.com/android-rpi/device_arpi_rpi4/commit/8b86126cb435be527c24936a623b31d5e1bf0fc0#diff-4509beb0ab401d71fa4a5cd94a55c9a74f13332776ae4019c5bfc4c2005157ff). I've noticed that in arpi-10 branch file `boot/cmdline.txt` is missing line `androidboot.serialno=10000000abcd1234`
- I used single USB-C to USB cable (USB-C connected to RPi 4) for power and data transfer simultaneously.

Thank you for such a great project!

closed 2021-01-30 by dananas

Aniketh01, 2021-07-30:
Didn't wanted to open a new issue for the same one. 

I managed to make the changes as mentioned by @dananas in the above comment. But still `adb devices` doesn't return any connected device when I'm connect over USB C to USB cable. Could you please suggest a way to debug this to understand what's going on?