android-rpi/device_arpi_rpi4 #64: Problem booting up android-11.0.0_r33 on RPi4

krisnil, 2021-03-22:
I have followed the instructions to build an image and sd-card for Raspberry Pi 4 and the build was successful. 

When I try to boot up the RPi 4 it boots up but get stuck at the Android logo. Is this a known issue or is there any solution for this?

nguyenanhgiau, 2021-05-19:
Hi @krisnil 
Let try this project: https://github.com/nguyenanhgiau/local_manifests/tree/rpi4-a11-telephony

vronchetti, 2021-05-26:
@krisnil I am observing the same issue - have you by chance found a fix/workaround?

vronchetti, 2021-05-26:
possible duplicate of https://github.com/android-rpi/device_arpi_rpi4/issues/57

nguyenanhgiau, 2022-04-03:
Hi guys,
Could you upload the failure log from UART or describe more detail about this problem?

AravindhSenthil, 2022-08-20:
Any update on this boot up issue @peyo-hd  @KonstaT 
Please help