android-rpi/device_arpi_rpi4 #66: Bluetooth not working

owahltinez, 2021-04-10:
I built the image (Android 11) as per the instructions. It boots and outputs to HDMI well. It also recognizes my wireless keyboard (via USB dongle). But Bluetooth is not working. Is that expected, or did I do something wrong?

peyo-hd, 2021-04-10:
Did you apply followings?
https://github.com/android-rpi/device_arpi_rpi4/wiki/arpi-11-:-framework-patch#add-hw-reset-on-bluetooth-hci-startup_timer_expired
https://github.com/android-rpi/device_arpi_rpi4/wiki/arpi-11-:-framework-patch#disable-low-power-mode-of-bluetooth

owahltinez, 2021-04-11:
I did. I created [patch files](https://github.com/owahltinez/device_arpi_rpi4/tree/arpi-11-patches) and applied them using [this script](https://github.com/owahltinez/local_manifests/blob/arpi-11/apply_patches.sh). The resulting image file can be found [here](https://github.com/owahltinez/device_arpi_rpi4/releases/tag/2021-04-10).

After the image boots, I go to settings and navigate to the screen to add a new remote or device. I put my device into pairing mode, but the Raspberry Pi is unable to find it. I have tried with multiple devices, all of which work when I use the LineageOS image.

peyo-hd, 2021-04-11:
For arpi-11 build,  try using 5.4 kernel build guided by following README.
https://github.com/android-rpi/device_arpi_rpi4/blob/arpi-11/README
with arpi-5.4.y kernel source.
https://github.com/android-rpi/kernel_arpi/tree/arpi-5.4.y

arpi-5.10 kernel is unstable with arpi-11 AOSP.

owahltinez, 2021-04-11:
Thanks for the suggestion. I tried using the [arpi-5.4.y](https://github.com/owahltinez/kernel_manifest/releases/tag/5.4.y) kernel, and I'm still seeing the same issue — stuck at the screen searching for devices, none of my bluetooth devices can be found.

Can you confirm that pairing Bluetooth accessories works?

peyo-hd, 2021-04-11:
It works.
You could check logs discussed in https://github.com/android-rpi/device_arpi_rpi4/issues/25

diggybub, 2021-11-16:
I have successfully built arpi-11 and was able to get bluetooth scanning working just fine in my apps. I am having problems connecting to any BLE device with the following error (after calling connectGatt()):

[WARNING:bta_gattc_act.cc(1044)] bta_gattc_conn_cback: cif=5 connected=0 conn_id=0x0005 reason=0x0008

My app works fine on any other Android device but fails on this build.
Has anyone had issues with connecting to BLE devices on this build?

mikegapinski, 2023-02-22:
Hello @peyo-hd 

Did you ever dig deeper into what needs to be done in order for BLE to work? It does not work in your implementation, glodroid(+Tesla Android) and lineage roms. I've spent a few days trying to figure it out but it seems our proprietary blobs are missing some hci commands that are used in broadcom/libbt. Generic linux bluetooth implementations on Android are also not fully working. There is no need to disable BLE in order to start the bt service but pairing with those devices is not possible at all.

Best regards