android-rpi/device_arpi_rpi4 #70: Tablet version of Android

lornova, 2021-05-15:
Is there a way to build the tablet version of Android instead of the Android TV interface?

peyo-hd, 2021-05-16:
https://github.com/lohriialo/device_arpi_rpi4

lornova, 2021-05-17:
> https://github.com/lohriialo/device_arpi_rpi4

I've been able to build AOSP with the device modified by @lohriialo for the tablet UI (branch arpi-10 as I need to target Android 10).

It boots but then launcher3 crashes. I get the error "Quickstep keeps stopping" and on dmesg errors like these:

[  284.657682] binder: undelivered TRANSACTION_COMPLETE
[  284.663501] binder: undelivered transaction 286543, process died.

On the same AOSP tree the build for Android TV works well.

Anyone has been able to have it working?

Sulaiman2050, 2021-07-22:
Have you used 1.6GB on your system partition ?