android-rpi/device_arpi_rpi4 #71: Build faild

WaylonYeoh, 2021-05-28:
Starting ninja...
FAILED: ninja: 'out/target/product/rpi4/system/lib/libbluetooth.so', needed by 'out/target/product/rpi4/system/lib/libbluetooth_qti.so', missing and no known rule to make it
08:47:52 ninja failed with: exit status 1

#### failed to build some targets (39 seconds) ####

csimmonds, 2021-06-03:
I have the same problem. The error results from package BluetoothQtiSymlink, which is not relevant to the RPi build. A simple fix is to do this:

mv device/google/vrservices/xr/packages/BluetoothQtiSymlink/Android.mk device/google/vrservices/xr/packages/BluetoothQtiSymlink/Android.not-mk

WaylonYeoh, 2021-06-16:
Thanks for your suggestion.This commit will not build the module.Does the Bluetooth work as follow your suggestion?

WaylonYeoh, 2021-06-16:
> I have the same problem. The error results from package BluetoothQtiSymlink, which is not relevant to the RPi build. A simple fix is to do this:
> 
> mv device/google/vrservices/xr/packages/BluetoothQtiSymlink/Android.mk device/google/vrservices/xr/packages/BluetoothQtiSymlink/Android.not-mk
> 

Thanks for your suggestion.This commit will not build the module.Does the Bluetooth work as follow your suggestion?

csimmonds, 2021-06-16:
Correct, this module will not be built, but since it is for a Qualcom chip it does not matter for us. It will have no effect on whether Bluetooth works on your RPi4 (i.e. if it is not working, this is not the cause, please open a separate issue)

nguyenanhgiau, 2021-06-26:
If you want to run: make -jX without any argument, you can refer my way. This way doesn't affect the other build.

`diff --git a/xr/packages/BluetoothQtiSymlink/Android.mk b/xr/packages/BluetoothQtiSymlink/Android.mk
index b25835a..b2a8a7c 100644
--- a/xr/packages/BluetoothQtiSymlink/Android.mk
+++ b/xr/packages/BluetoothQtiSymlink/Android.mk
@@ -1,7 +1,7 @@
 #
 # This is a workaround for Bluetooth not working on OnePlus7 Pro. See b/139486342
 #
-
+ifeq ($(filter rpi4, $(TARGET_DEVICE)),)
 LOCAL_PATH := $(call my-dir)
 
 include $(CLEAR_VARS)
@@ -18,3 +18,4 @@ $(bluetooth_qti_path): $(bluetooth_path)
        cd $(lib_dir) && ln -sf $(bluetooth) $(bluetooth_qti)
 
 droid: $(bluetooth_qti_path)
+endif
\ No newline at end of file`

ranvijay-tss, 2024-10-24:
Thanks Man, For the solution.