android-rpi/device_arpi_rpi4 #73: Build error in branch arpi-12

MengMengDaXiaoJi, 2021-06-30:
I just do this :

source build/envsetup.sh
lunch rpi4-eng
make ramdisk systemimage vendorimage

But error is:

FAILED: //frameworks/base:framework-minus-apex javac18
Outputs: out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/framework-minus-apex.jar18
Error: exited with code: 1
Command: rm -rf "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/classes" "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/anno" "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/srcjars" "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/framework-minus-apex.jar18" && mkdir -p "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/classes" "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/anno" "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/srcjars" && out/soong/host/linux-x86/bin/zipsync -d out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/srcjars -l out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/srcjars/list -f "*.java"  && (if [ -s out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/srcjars/list ] || [ -s out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/framework-minus-apex.jar18.rsp ] ; then out/soong/host/linux-x86/bin/soong_javac_wrapper prebuilts/jdk/jdk11/linux-x86/bin/javac -J-Xmx2048M -J-XX:OnError="cat hs_err_pid%p.log" -J-XX:CICompilerCount=6 -J-XX:+UseDynamicNumberOfGCThreads -Xmaxerrs 9999999 -encoding UTF-8 -sourcepath "" -g -XDskipDuplicateBridges=true -XDstringConcat=inline -processorpath out/soong/.intermediates/tools/platform-compat/java/android/processor/compat/changeid/compat-changeid-annotation-processor/linux_glibc_common/withres/compat-changeid-annotation-processor.jar:out/soong/.intermediates/tools/platform-compat/java/android/processor/compat/unsupportedappusage/unsupportedappusage-annotation-processor/linux_glibc_common/withres/unsupportedappusage-annotation-processor.jar:out/soong/.intermediates/frameworks/base/tools/processors/view_inspector/view-inspector-annotation-processor/linux_glibc_common/withres/view-inspector-annotation-processor.jar:out/soong/.intermediates/frameworks/base/tools/processors/staledataclass/staledataclass-annotation-processor/linux_glibc_common/withres/staledataclass-annotation-processor.jar -processor android.processor.compat.changeid.ChangeIdProcessor,android.processor.compat.unsupportedappusage.UnsupportedAppUsageProcessor,android.processor.view.inspector.PlatformInspectableProcessor,android.processor.staledataclass.StaleDataclassProcessor -Xlint:-dep-ann --system=out/soong/.intermediates/libcore/mmodules/core_platform_api/legacy-core-platform-api-stubs-system-modules/android_common/system -classpath out/soong/.intermediates/tools/platform-compat/java/android/compat/annotation/app-compat-annotations/android_common/turbine-combined/app-compat-annotations.jar:out/soong/.intermediates/frameworks/base/ext/android_common/turbine-combined/ext.jar:out/soong/.intermediates/frameworks/base/framework-updatable-stubs-module_libs_api/android_common/turbine-combined/framework-updatable-stubs-module_libs_api.jar:out/soong/.intermediates/tools/platform-compat/java/android/compat/annotation/unsupportedappusage/android_common/turbine-combined/unsupportedappusage.jar:out/soong/.intermediates/external/bouncycastle/bouncycastle-repackaged-unbundled/android_common/turbine-combined/bouncycastle-repackaged-unbundled.jar:out/soong/.intermediates/frameworks/base/framework-internal-utils/android_common/turbine-combined/framework-internal-utils.jar:out/soong/.intermediates/frameworks/base/mime/mimemap/android_common/turbine-combined/mimemap.jar:out/soong/.intermediates/frameworks/libs/modules-utils/java/com/android/modules/utils/modules-utils-os/android_common/turbine-combined/modules-utils-os.jar:out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/turbine-combined/framework-minus-apex.jar -source 1.9 -target 1.9 -d out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/classes -s out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/anno @out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/framework-minus-apex.jar18.rsp @out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/srcjars/list ; fi ) && out/soong/host/linux-x86/bin/soong_zip -jar -o out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/framework-minus-apex.jar18 -C out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/classes -D out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/classes && rm -rf "out/soong/.intermediates/frameworks/base/framework-minus-apex/android_common/javac/shard18/srcjars"
Output:
[1mframeworks/base/core/java/com/android/internal/os/WrapperInit.java:24: [31merror:[0m[1m cannot find symbol[0m
import android.system.StructUserCapData;
[1m                     [32m^[0m[1m[0m
  symbol:   class StructUserCapData
  location: package android.system
[1mframeworks/base/core/java/com/android/internal/os/WrapperInit.java:25: [31merror:[0m[1m cannot find symbol[0m
import android.system.StructUserCapHeader;
[1m                     [32m^[0m[1m[0m
  symbol:   class StructUserCapHeader
  location: package android.system
[1mframeworks/base/core/java/com/android/internal/os/ZygoteInit.java:48: [31merror:[0m[1m cannot find symbol[0m
import android.system.StructUserCapData;
[1m                     [32m^[0m[1m[0m
  symbol:   class StructUserCapData
  location: package android.system
[1mframeworks/base/core/java/com/android/internal/os/ZygoteInit.java:49: [31merror:[0m[1m cannot find symbol[0m
import android.system.StructUserCapHeader;
[1m                     [32m^[0m[1m[0m
  symbol:   class StructUserCapHeader
  location: package android.system
4 errors

nguyenanhgiau, 2021-07-01:
Hi @MengMengDaXiaoJi
My solution is reverting these files. This is my patch for frameworks/base.

```
diff --git a/core/java/com/android/internal/os/WrapperInit.java b/core/java/com/android/internal/os/WrapperInit.java
index 508782b9f3dc..c6cd5350a589 100644
--- a/core/java/com/android/internal/os/WrapperInit.java
+++ b/core/java/com/android/internal/os/WrapperInit.java
@@ -21,8 +21,8 @@ import android.os.Trace;
 import android.system.ErrnoException;
 import android.system.Os;
 import android.system.OsConstants;
-import android.system.StructUserCapData;
-import android.system.StructUserCapHeader;
+import android.system.StructCapUserData;;
+import android.system.StructCapUserHeader;
 import android.util.Slog;
 import android.util.TimingsTraceLog;
 
@@ -187,9 +187,9 @@ public class WrapperInit {
      *       capabilities, which may make it crash, but not exceed its allowances.
      */
     private static void preserveCapabilities() {
-        StructUserCapHeader header = new StructUserCapHeader(
+        StructCapUserHeader header = new StructCapUserHeader(
                 OsConstants._LINUX_CAPABILITY_VERSION_3, 0);
-        StructUserCapData[] data;
+        StructCapUserData[] data;
         try {
             data = Os.capget(header);
         } catch (ErrnoException e) {
@@ -199,9 +199,9 @@ public class WrapperInit {
 
         if (data[0].permitted != data[0].inheritable ||
                 data[1].permitted != data[1].inheritable) {
-            data[0] = new StructUserCapData(data[0].effective, data[0].permitted,
+            data[0] = new StructCapUserData(data[0].effective, data[0].permitted,
                     data[0].permitted);
-            data[1] = new StructUserCapData(data[1].effective, data[1].permitted,
+            data[1] = new StructCapUserData(data[1].effective, data[1].permitted,
                     data[1].permitted);
             try {
                 Os.capset(header, data);
diff --git a/core/java/com/android/internal/os/ZygoteInit.java b/core/java/com/android/internal/os/ZygoteInit.java
index 0121b789f39a..7eb030666d1c 100644
--- a/core/java/com/android/internal/os/ZygoteInit.java
+++ b/core/java/com/android/internal/os/ZygoteInit.java
@@ -45,8 +45,8 @@ import android.security.keystore2.AndroidKeyStoreProvider;
 import android.system.ErrnoException;
 import android.system.Os;
 import android.system.OsConstants;
-import android.system.StructUserCapData;
-import android.system.StructUserCapHeader;
+import android.system.StructCapUserData;;
+import android.system.StructCapUserHeader;;
 import android.text.Hyphenator;
 import android.util.EventLog;
 import android.util.Log;
@@ -742,9 +742,9 @@ public class ZygoteInit {
                 OsConstants.CAP_BLOCK_SUSPEND
         );
         /* Containers run without some capabilities, so drop any caps that are not available. */
-        StructUserCapHeader header = new StructUserCapHeader(
+        StructCapUserHeader header = new StructCapUserHeader(
                 OsConstants._LINUX_CAPABILITY_VERSION_3, 0);
-        StructUserCapData[] data;
+        StructCapUserData[] data;
         try {
             data = Os.capget(header);
         } catch (ErrnoException ex) {

```

MengMengDaXiaoJi, 2021-07-06:
@nguyenanhgiau
Thank you very much. I have solved it by using your patch.

closed 2021-07-06 by MengMengDaXiaoJi