android-rpi/device_arpi_rpi4 #76: Failure [INSTALL_FAILED_NO_MATCHING_ABIS: Failed to extract native libraries, res=-113]

Aniketh01, 2021-07-27:
When I try to install apps through adb, I get an error like this: adb: failed to install app.apk: Failure [INSTALL_FAILED_NO_MATCHING_ABIS: Failed to extract native libraries, res=-113]. What is the root cause of this error? How can I fix this?

peyo-hd, 2021-07-28:
arpi-11, arpi-12 builds are supporting Arm 64bit ABI only.

Aniketh01, 2021-07-28:
Thanks for the response. Is there any commits or references that I can follow to add multi ABI support?

peyo-hd, 2021-07-28:
https://github.com/android-rpi/device_arpi_rpi4/commit/fe2c6e7a42fde5ed95344636cbaeb839aa953db4

Aniketh01, 2021-07-28:
Can you please elaborate a bit more? What sort of issues might I face?

Aniketh01, 2021-07-30:
Have you tested AOSP 11 for 32 bit ABI alone without multilib support? Does it work well?

closed 2021-08-05 by Aniketh01