android-rpi/device_arpi_rpi4 #81: thanks mr. peyo-hd! how to build andorid-s-beta4?

andersling, 2021-09-01:
Hi,MR.peyo-hd ! Thanks for your greate job! !  
  and when i build android-s-beta 4 which is the newest version released by google. I found that it failed. 
I use ubuntu 20.04 LTS ,desktop version.follow your instruction,but i replaced android-s-beta2 with android-s-beta-4.
when i run the build commands ,it failed:
     
============================================
PLATFORM_VERSION_CODENAME=T
PLATFORM_VERSION=T
TARGET_PRODUCT=rpi4
TARGET_BUILD_VARIANT=eng
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a72
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.11.0-27-generic-x86_64-Ubuntu-20.04.3-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=AOSP.MASTER
OUT_DIR=out
PRODUCT_SOONG_NAMESPACES=external/mesa3d
============================================

============================================
[100% 1/1] out/soong/.bootstrap/bin/soong_build out/soong/build.ninja
FAILED: out/soong/build.ninja
cd "$(dirname "out/soong/.bootstrap/bin/soong_build")" && BUILDER="$PWD/$(basename "out/soong/.bootstrap/bin/soong_build")" && cd / && 
env -i "$BUILDER"     --top "$TOP"     --out "out/soong"     -n "out"     -d "out/soong/build.ninja.d"     -t -l out/.module_paths/Andr
oid.bp.list -globFile out/soong/.bootstrap/build-globs.ninja -o out/soong/build.ninja --available_env out/soong/soong.environment.avail
able --used_env out/soong/soong.environment.used Android.bp
error: external/rust/crates/zip/Android.bp:21:1: "libzip" depends on undefined module "libflate2"
16:11:03 soong bootstrap failed with: exit status 1
ninja: build stopped: subcommand failed.

any idea? thanks@!!!

peyo-hd, 2021-09-02:
I also failed to build s-beta4, then did not update the version.

andersling, 2021-09-02:
> I also failed to build s-beta4, then did not update the version.

Thanks fro your reply, mr. peyo-hd!
but google say that beta 4 is almost the stability version privous to the release version, so I wish someone can get beta4 to work with rpi4.

Thanks for your greate jobs again!! Have a nice day!