android-rpi/device_arpi_rpi4 #92: Click developer options  just come back to home,no response

andersling, 2021-12-19:
hi, MR.peyo-hd ! Big thanks for your greate job ! This release the WifFI ,bluethooth are worked well!

My device are Rpi4 8G and CM4 4G:
but I has found 2 issues:
    1. the start aminmation of "Android" is flicking.
    2. The developer options is not functional, when I clicked on it ,just come back to Home screen!

 pls have a look at these issues, thanks,sir!

andersling, 2021-12-20:
I get the debug message from ttyUSB0 is :

>**  [54808.314142] binder: undelivered transaction 166130, process died.
    [54808.320972] binder: undelivered transaction 166121, process died.
    [54845.136100] binder_alloc: 3331: binder_alloc_buf, no vma
    [54845.138624] binder: undelivered transaction 172896, process died.
    [54845.141803] binder: 249:249 transaction failed 29189/-3, size 320-8 line 2917**

andersling, 2021-12-20:
@peyo-hd  pls help,thanks!

AravindhSenthil, 2022-08-19:
@andersling 

Any update you have on this?

andersling, 2022-08-21:
> 

This issue in no longer exists in my room now ,maybe i sync the latest source from this repo and rebuild!

Good luck to you!