android-rpi/device_arpi_rpi4 #93: Sometimes boot failed: black screen, and sometime fleeking with andorid animation and background like terminal

andersling, 2021-12-26:
hi,mr.peyo-hd:
   I found another issue: sometimes the boot process is frozen, just black screen. and the Message found with the ttyUSB0 are:

  

> <pre>[    9.685017] type=1400 audit(9.679:8): avc: denied { read } for comm=&quot;android.hardwar&quot; name=&quot;u:object_r:default_prop:s0&quot; dev=&quot;tmpfs&quot; ino=94 scontext=u:r:hal_graphics_allocator_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=1
[   10.359301] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   14.365576] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   14.480911] v3d fec00000.v3d: MMU error from client ? (13) at 0x191c180, pte invalid
[   15.367975] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   16.370667] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   17.372750] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   18.374338] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   19.375570] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   19.915098] type=1400 audit(19.911:16): avc: denied { read } for comm=&quot;Binder:189_2&quot; name=&quot;wakeup0&quot; dev=&quot;sysfs&quot; ino=14570 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   19.934400] type=1400 audit(19.911:17): avc: denied { open } for comm=&quot;Binder:189_2&quot; path=&quot;/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0&quot; dev=&quot;sysfs&quot; ino=14570 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   19.960801] type=1400 audit(19.911:18): avc: denied { read } for comm=&quot;Binder:189_2&quot; name=&quot;event_count&quot; dev=&quot;sysfs&quot; ino=14577 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[   19.980107] type=1400 audit(19.911:19): avc: denied { open } for comm=&quot;Binder:189_2&quot; path=&quot;/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0/event_count&quot; dev=&quot;sysfs&quot; ino=14577 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[   20.007722] type=1400 audit(19.911:20): avc: denied { getattr } for comm=&quot;Binder:189_2&quot; path=&quot;/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/wakeup/wakeup0/event_count&quot; dev=&quot;sysfs&quot; ino=14577 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=file permissive=1
[   20.375924] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   21.378235] init: Control message: Could not find &apos;aidl/package_native&apos; for ctl.interface_start from pid: 171 (/system/bin/servicemanager)
[   21.952102] init: Control message: Processed ctl.start for &apos;idmap2d&apos; from pid: 380 (system_server)
[   21.996763] healthd: battery none chg=
[   22.461764] type=1400 audit(1639829003.263:21): avc: denied { read } for comm=&quot;InputReader&quot; name=&quot;leds&quot; dev=&quot;sysfs&quot; ino=3972 scontext=u:r:system_server:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   22.480723] type=1400 audit(1639829003.263:22): avc: denied { open } for comm=&quot;InputReader&quot; path=&quot;/sys/devices/platform/leds&quot; dev=&quot;sysfs&quot; ino=3972 scontext=u:r:system_server:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=1
[   31.982038] init: Sending signal 9 to service &apos;idmap2d&apos; (pid 259) process group...
[   31.998588] libprocessgroup: Successfully killed process cgroup uid 1000 pid 259 in 8ms
[   32.008448] init: Control message: Processed ctl.stop for &apos;idmap2d&apos; from pid: 380 (system_server)
[   32.018205] init: Service &apos;idmap2d&apos; (pid 259) received signal 9

</pre>

 would you do me a favor have a look at it?
 big thanks! @peyo-hd @KonstaT @ikalkov

peyo-hd, 2022-01-20:
Try build with android-12.0.0_r27.
Recently,  I dropped drm_gralloc.
And graphics hal directly uses mesa gbm.

andersling, 2022-01-21:
> peyo-hd

thanks a lot,MR.peyo-hd! You are amazing: make RPI4 dance with AOSP 12!!

mse6cob, 2022-03-09:
@peyo-hd @andersling 

I have followed the steps mentioned in the below link. Build is successful( Ramdisk, system and vendor image is getting generated)

https://github.com/android-rpi/local_manifests

But I got the below error while building the kernel with arpi-5.10.

https://github.com/android-rpi/kernel_manifest

$ build/build.sh
build/build.sh: line 424: nproc: command not found
build/build.sh: line 425: readlink: command not found
build/build.sh: line 426: readlink: command not found
build/build.sh: line 427: readlink: command not found
build/build.sh: line 472: rm: command not found

Because of that, I used the below kernel manifest to build the **kernel 5.10**.

https://github.com/MaJiu/kernel_manifest

Kernel build also successful and created the partition in USB instead of SD-Card as mentioned in the below link.

https://github.com/android-rpi/device_arpi_rpi4

**Raspberry Pi goes into continuous reboot.**

Kindly let me know how to get android12 running in Pi board or if I have skipped any steps.

**Note:** I have already tried with android-11 via USB and it was working in raspberry pi.

Thanks.