android-rpi/device_arpi_rpi4 #96: Build kernel fatal error: 'sys/types.h' file not found ???

Sera5, 2022-01-25:
#
+ set +x
+ cd common
+ make LLVM=1 LLVM_IAS=1 DEPMOD=depmod DTC=dtc O=/home/sera/holo_kernel/out/arpi-5.10/common arpi_defconfig
make[1]: Verzeichnis „/home/sera/holo_kernel/out/arpi-5.10/common“ wird betreten
  GEN     Makefile
  HOSTCC  scripts/basic/fixdep
/home/sera/holo_kernel/common/scripts/basic/fixdep.c:92:10: fatal error: 'sys/types.h' file not found
#include <sys/types.h>
         ^~~~~~~~~~~~~
1 error generated.
make[2]: *** [scripts/Makefile.host:95: scripts/basic/fixdep] Fehler 1
make[1]: *** [/home/sera/holo_kernel/common/Makefile:550: scripts_basic] Fehler 2
make[1]: Verzeichnis „/home/sera/holo_kernel/out/arpi-5.10/common“ wird verlassen
make: *** [Makefile:185: __sub-make] Error 2

sera@sera-System-Product-Name:~/holo_kernel$ find /usr/include -name types.h
/usr/include/rpc/types.h
/usr/include/x86_64-linux-gnu/asm/types.h
/usr/include/x86_64-linux-gnu/bits/types.h
/usr/include/x86_64-linux-gnu/sys/types.h
/usr/include/linux/iio/types.h
/usr/include/linux/types.h
/usr/include/linux/sched/types.h
/usr/include/asm-generic/types.h
/usr/include/sys/types.h

ElectroBoy404NotFound, 2022-01-29:
it is working for me, try to reclone/redownload the repo

EinarArnason, 2022-02-03:
I have this problem as well out of the box.
Also when I try to build rpi aosp out of the box I get the error below. Seems something has changed in environment in newer distros.

`FAILED: src/gallium/auxiliary/libgallium.a.p/hud_hud_sensors_temp.c.o 
/home/einar/systemat/travelboy-aosp/prebuilts/clang/host/linux-x86/clang-r416183b1/bin/clang '' -Isrc/gallium/auxiliary/libgallium.a.p -Isrc/gallium/auxiliary -I../src/gallium/auxiliary -Isrc/loader -I../src/loader -I../src/gallium/include -Isrc -I../src -Iinclude -I../include -I../src/gallium/auxiliary/util -Isrc/compiler/nir -I../src/compiler/nir -Isrc/util -I../src/util -I/home/einar/systemat/travelboy-aosp/external/mesa3d/android -I/home/einar/systemat/travelboy-aosp/out/target/product/rpi4/obj/SHARED_LIBRARIES/meson.dummy.64_intermediates -I/home/einar/systemat/travelboy-aosp/out/target/product/rpi4/gen/SHARED_LIBRARIES/meson.dummy.64_intermediates -I/home/einar/systemat/travelboy-aosp/external/libdrm/include/drm -I/home/einar/systemat/travelboy-aosp/external/libdrm/android -I/home/einar/systemat/travelboy-aosp/system/logging/liblog/include_vndk -I/home/einar/systemat/travelboy-aosp/system/core/libcutils/include_outside_system -I/home/einar/systemat/travelboy-aosp/system/core/libprocessgroup/include -I/home/einar/systemat/travelboy-aosp/system/core/libcutils/include -I/home/einar/systemat/travelboy-aosp/external/zlib -I/home/einar/systemat/travelboy-aosp/external/libcxx/include -I/home/einar/systemat/travelboy-aosp/external/libcxxabi/include -I/home/einar/systemat/travelboy-aosp/frameworks/native/libs/nativewindow/include -I/home/einar/systemat/travelboy-aosp/system/core/libsync/include -I/home/einar/systemat/travelboy-aosp/hardware/libhardware/include -I/home/einar/systemat/travelboy-aosp/system/media/audio/include -I/home/einar/systemat/travelboy-aosp/system/core/libsystem/include -I/home/einar/systemat/travelboy-aosp/system/bt/types -I/home/einar/systemat/travelboy-aosp/system/libhidl/base/include -I/home/einar/systemat/travelboy-aosp/system/libhidl/transport/include -I/home/einar/systemat/travelboy-aosp/system/libfmq/base -I/home/einar/systemat/travelboy-aosp/system/libhwbinder/include -I/home/einar/systemat/travelboy-aosp/system/libbase/include -I/home/einar/systemat/travelboy-aosp/external/fmtlib/include -I/home/einar/systemat/travelboy-aosp/system/core/libutils/include -I/home/einar/systemat/travelboy-aosp/system/unwinding/libbacktrace/include -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/system/libhidl/transport/manager/1.0/android.hidl.manager@1.0_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/system/libhidl/transport/manager/1.1/android.hidl.manager@1.1_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/system/libhidl/transport/manager/1.2/android.hidl.manager@1.2_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/system/libhidl/transport/base/1.0/android.hidl.base@1.0_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/hardware/interfaces/graphics/common/1.0/android.hardware.graphics.common@1.0_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/hardware/interfaces/graphics/common/1.1/android.hardware.graphics.common@1.1_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/hardware/interfaces/graphics/common/1.2/android.hardware.graphics.common@1.2_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/hardware/interfaces/graphics/mapper/4.0/android.hardware.graphics.mapper@4.0_genc++_headers/gen -I/home/einar/systemat/travelboy-aosp/frameworks/native/libs/gralloc/types/include -I/home/einar/systemat/travelboy-aosp/frameworks/native/libs/binder/ndk/include_cpp -I/home/einar/systemat/travelboy-aosp/frameworks/native/libs/binder/ndk/include_ndk -I/home/einar/systemat/travelboy-aosp/frameworks/native/libs/binder/ndk/include_platform -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/hardware/interfaces/common/aidl/android.hardware.common-V2-ndk_platform-source/gen/include -I/home/einar/systemat/travelboy-aosp/out/soong/.intermediates/hardware/interfaces/graphics/common/aidl/android.hardware.graphics.common-V2-ndk_platform-source/gen/include -I/home/einar/systemat/travelboy-aosp/external/expat/lib -I/home/einar/systemat/travelboy-aosp/frameworks/native/libs/arect/include -I/home/einar/systemat/travelboy-aosp/external/elfutils/libelf -I/home/einar/systemat/travelboy-aosp/external/elfutils -I/home/einar/systemat/travelboy-aosp/external/elfutils/include -I/home/einar/systemat/travelboy-aosp/external/elfutils/lib -I/home/einar/systemat/travelboy-aosp/external/elfutils/bionic-fixup -I/home/einar/systemat/travelboy-aosp/frameworks/native/libs/nativebase/include -I/home/einar/systemat/travelboy-aosp/frameworks/native/vulkan/include -fvisibility=hidden -fcolor-diagnostics -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c11 -O3 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS '-DPACKAGE_VERSION="21.3.4"' '-DPACKAGE_BUGREPORT="https://gitlab.freedesktop.org/mesa/mesa/-/issues"' -DUSE_ELF_TLS -DENABLE_ST_OMX_BELLAGIO=0 -DENABLE_ST_OMX_TIZONIA=0 -DEGL_NO_X11 -DHAVE_ANDROID_PLATFORM -DANDROID_API_LEVEL=31 -DENABLE_SHADER_CACHE -DHAVE___BUILTIN_BSWAP32 -DHAVE___BUILTIN_BSWAP64 -DHAVE___BUILTIN_CLZ -DHAVE___BUILTIN_CLZLL -DHAVE___BUILTIN_CTZ -DHAVE___BUILTIN_EXPECT -DHAVE___BUILTIN_FFS -DHAVE___BU [... cut ...] i4/obj/include '' -fPIC -pthread -Werror=pointer-arith -Werror=gnu-empty-initializer -MD -MQ src/gallium/auxiliary/libgallium.a.p/hud_hud_sensors_temp.c.o -MF src/gallium/auxiliary/libgallium.a.p/hud_hud_sensors_temp.c.o.d -o src/gallium/auxiliary/libgallium.a.p/hud_hud_sensors_temp.c.o -c ../src/gallium/auxiliary/hud/hud_sensors_temp.c
../src/gallium/auxiliary/hud/hud_sensors_temp.c:47:10: fatal error: 'sensors/sensors.h' file not found
#include <sensors/sensors.h>
         ^~~~~~~~~~~~~~~~~~~`

ElectroBoy404NotFound, 2022-02-03:
> I have this problem as well out of the box. Also when I try to build rpi aosp out of the box I get the error below. Seems something has changed in environment in newer distros.
> 
> `../src/gallium/auxiliary/hud/hud_sensors_temp.c:47:10: fatal error: 'sensors/sensors.h' file not found #include <sensors/sensors.h> ^~~~~~~~~~~~~~~~~~~`

I guess something is wrong with your build environment, i think i was able to build this as i already had a build system setup that was working

EinarArnason, 2022-02-03:
I'm curious to know what it might be. Vanilla aosp builds successfully and these header files exist. So this is something specific.

EinarArnason, 2022-02-06:
I have lmsensors installed and explicitly disabling lmsensors in meson build in the mesa3d project solved that issue. About the kernel project, I setup common kernel source first and added rpi specific projects as local manifests and the build completes.

MaJiu, 2022-02-23:
> * set +x
> * cd common
> * make LLVM=1 LLVM_IAS=1 DEPMOD=depmod DTC=dtc O=/home/sera/holo_kernel/out/arpi-5.10/common arpi_defconfig
>   make[1]: Verzeichnis „/home/sera/holo_kernel/out/arpi-5.10/common“ [...]

Hi,  Sera5. Did you solve this problem? I had the same problem as you, but I solved it now. You can try my kernel manifest(**https://github.com/MaJiu/kernel_manifest**), I made some changes. (ref : https://android.googlesource.com/kernel/manifest/+/ 1780a51cda4a6819a558d047b903476c8bd977b5, https://android.googlesource.com/kernel/manifest/+/ b25f5e4631790b1c3cdc1b1ee2e23f519eabe37e).  

I guess it's because the branch has changed.
> fix superproject definition: include revision
> 
> repo changed the default handling for superproject. It no longer
> considers the containing branch name, but rather the default revision.
> Mitigate that by explicitly setting the revision in the superproject
> tag.