android-rpi/device_arpi_rpi4 #101: rpi-11 boot issue.

pradeep746, 2022-07-18:
I have rpi -4B with 8gb module when i am trying to flash source code it's showing start4.elf is not compatible this board requires newer software.

gogamid, 2022-08-14:
@pradeep746  take sd card and plug it to your laptop. Then download `boot/fixup4.dat `and` boot/start4.elf` from [here](https://github.com/android-rpi/device_arpi_rpi4/commit/c686a27cf0c138f20aa480f009c2b49bcc516c09). Then replace those files in sd card with downloaded versions. That's it, now you can plug sd card to the rpi4, it will boot.

pradeep746, 2022-08-22:
@gogamid 
I have tried this process but it is sucking on the boot-loader animation.

gogamid, 2022-08-23:
@pradeep746 it seems some packages are missing, try to connect to rpi via serial console. Guide:

pradeep746, 2022-08-23:
@gogamid I have update those file then it's showing rainbow pic and continue restarting.

gogamid, 2022-08-23:
@pradeep746   I forked, but did not do any modifications, so do not fork it :) 
So did you try to connect via serial and see the logs?

appapurapu, 2024-03-28:
The RPI4 boot works fine after replacing the start4.elf and fixup4.dat with the the latest git copies