android-rpi/device_arpi_rpi4 #106: Build Android 12 error

khoaggle, 2022-08-22:
****
This happen when i reached 91% 
Does anyone know how to fix it ?
Below is the log 

[ 91% 73633/80492] build out/target/product/rpi4/obj/MESON_MESA3D/.build.timesta
FAILED: out/target/product/rpi4/obj/MESON_MESA3D/.build.timestamp
/bin/bash -c "(rm -rf out/target/product/rpi4/obj/MESON_MESA3D/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_MESA3D/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_MESA3D//build ) && (cp -r external/mesa3d//* out/target/product/rpi4/obj/MESON_MESA3D/ ) && (cd out/target/product/rpi4/obj/MESON_MESA3D && PATH=/usr/bin:/usr/local/bin:\$PATH meson ./build --cross-file /home/guk81hc/samba/views/android12/out/target/product/rpi4/obj/MESON_MESA3D_GEN/aosp_cross --buildtype=release -Ddri-search-path=/vendor/lib64/dri -Dplatforms=android -Dplatform-sdk-version=32 -Ddri-drivers= -Dgallium-drivers=v3d,vc4 -Dvulkan-drivers=broadcom -Dgbm=enabled -Degl=enabled -Dcpp_rtti=false ) && (PATH=/usr/bin:/bin:/sbin:\$PATH ninja -C out/target/product/rpi4/obj/MESON_MESA3D/build ) && (touch out/target/product/rpi4/obj/MESON_MESA3D/.build.timestamp )"
The Meson build system
Version: 0.45.1
Source dir: /home/guk81hc/samba/views/android12/out/target/product/rpi4/obj/MESON_MESA3D
Build dir: /home/guk81hc/samba/views/android12/out/target/product/rpi4/obj/MESON_MESA3D/build
Build type: cross build

**meson.build:485:9: ERROR: Expecting eol got id.
if 'x11' in _platforms
         ^**

A full log can be found at /home/guk81hc/samba/views/android12/out/target/product/rpi4/obj/MESON_MESA3D/build/meson-logs/meson-log.txt
14:38:49 ninja failed with: exit status 1

#### failed to build some targets (01:35:56 (hh:mm:ss)) ####

khoaggle, 2022-08-22:
@peyo-hd  please help me

gogamid, 2022-08-23:
@khoaggle I had same issue. 
first make sure you have python3.8 as default for python or python3 command, you can do that by running these commands:

```
#configure python
sudo apt install libssl-dev python3-setuptools
#install python3.8 and set as default
sudo apt install python3.8
sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.6 1
sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.8 2
sudo update-alternatives --config python3
#choose 2 (python3.8) as default
#link python to python3
sudo ln -s /usr/bin/python3 /usr/bin/python
```

Then you can update meson

```
#update meson
sudo apt-get remove meson
sudo apt-get install python3 python3-pip python3-setuptools \
                       python3-wheel ninja-build
pip3 install --user meson
#move meson from local to root
sudo mv .local/bin/meson /usr/bin/
source .profile
```

khoaggle, 2022-08-23:
> @khoaggle I had same issue. first make sure you have python3.8 as default for python or python3 command, you can do that by running these commands:
>
> ```
> #configure python
> sudo apt install libssl-dev [...]

khoaggle, 2022-08-23:
Thank you a lot you just saved my life

gogamid, 2022-08-23:
you are welcome