android-rpi/device_arpi_rpi4 #111: Could not allocate block in ext2 filesystem

flug, 2022-12-29:
I'm trying to compile an "automative os" version, the command ``make ramdisk systemimage vendorimage`` ends up crashing with this error: 

__populate_fs: Unable to allocate a block in the ext2 filesystem when writing the "webview.apk" file.
e2fsdroid: Unable to allocate a block in an ext2 file system when creating a file system.

The build is done via a docker image based on Ubuntu, at the beginning I had started this same build on my OS (Linux manjaro) and I had the same error. I don't quite understand how I could solve this problem? Is it due to my partition on my pc which is in est4 which makes ext2 not work?

alsutton, 2023-02-01:
From what I've seen, this usually happens when you've added things to the build which can not fit into the size allocated for one of the partitions (usually `system.img` or `vendor.img`). If you have added anything, try removing it, and seeing if the build finishes.

FlorianHeigl, 2026-11-04:
i had this when building a different image inside a docker container on zfs, see https://c-goes.github.io/posts/bitbake-yocto-could-not-allocate-block-in-ext2-filesystem/