android-rpi/device_arpi_rpi4 #113: Android-13.0.0_r23 Cannot find framework matrix at FCM version 2.

aladin153, 2023-02-07:
I' trying to build Android 13 r23 for rpi4 and I'm facing the following issue :
`ERROR: No such file or directory: #Cannot find framework matrix at FCM version 2.: No such file or directory`

The solution was to modify the target-level in manifest file from 2 to 3.
`-<manifest version="1.0" type="device"  #target-level="2">
 +<manifest version="1.0" type="device" target-level="3">
`

[Fail.txt](https://github.com/android-rpi/device_arpi_rpi4/files/10680393/Fail.txt)

vnguyentrong, 2023-12-05:
Hello @aladin153, have you fixed this issue?, if yes, could you please summarize how to fix this?

AhnMo, 2024-01-29:
Build with following command:
```
$ make ramdisk systemimage vendorimage
```
not this one:
```
$ m
```
https://github.com/android-rpi/device_arpi_rpi4/blob/arpi-13/README#L13

0mew, 2024-08-07:
It's giving an error now. What should I do to solve it? Have you examined the manifest, where exactly is it?

`  android.hardware.memtrack@1.0::IMemtrack/default
  android.hardware.nfc@1.2::INfc/default
  android.hardware.power.IPower/default (@4)
  android.hardware.sensors@1.0::ISensors/default
  android.hardware.soundtrigger@2.2::ISoundTriggerHw/default
  android.hardware.tetheroffload.config@1.0::IOffloadConfig/default
  android.hardware.tetheroffload.control@1.0::IOffloadControl/default
  android.hardware.wifi.IWifi/default (@2)
  android.hardware.wifi.hostapd.IHostapd/default (@2)
  android.hardware.wifi.supplicant.ISupplicant/default (@3)
ERROR: No such file or directory: Cannot find framework matrix at FCM version 4.: No such f
ile or directory
14:13:32 ninja failed with: exit status 1

`