android-rpi/device_arpi_rpi4 #116: Compilation error android_hardware_Camera.cpp on Ubuntu

mihaimacarie98, 2023-04-10:
[[  3% 1055/27872] //frameworks/base/core/jni:libandroid_runtime clang++ android_hardware_Camera.cpp
FAILED: out/soong/.intermediates/frameworks/base/core/jni/libandroid_runtime/android_arm64_armv8-a_cortex-a72_shared/obj/frameworks/base/core/jni/android_hardware_Camera.o
PWD=/proc/self/cwd prebuilts/clang/host/linux-x86/clang-r450784d/bin/clang++ -c  -Werror=implicit-function-declaration -DANDROID -fmessage-length=0 -W -Wall -Wno-unused -Winit-self -Wpointer-arith -Wunreachable-code-loop-increment -no-canonical-prefixes -DNDEBUG -UDEBUG -fno-exceptions -Wno-multichar -O2 -g -fdebug-default-version=5 -fno-strict-aliasing -Werror=date-time -Werror=pragma-pack -Werror=pragma-pack-suspicious-include -Werror=string-plus-int -Werror=unreachable-code-loop-increment -D__compiler_offsetof=__builtin_offsetof -faddrsig -fcommon -Werror=int-conversion -fexperimental-new-pass-manager -Wno-reserved-id-macro -fcolor-diagnostics -Wno-sign-compare -Wno-defaulted-function-deleted -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-gnu-designator -Wno-gnu-folding-constant -Wunguarded-availability -D__ANDROID_UNAVAILABLE_SYMBOLS_ARE_WEAK__ -ffp-contract=off -fdebug-prefix-map=/proc/self/cwd= -ftrivial-auto-var-init=zero -enable-trivial-auto-var-init-zero-knowing-it-will-be-removed-from-clang -ffunction-sections -fdata-sections -fno-short-enums -funwind-tables -fstack-protector-strong -Wa,--noexecstack -D_FORTIFY_SOURCE=2 -Wstrict-aliasing=2 -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Werror=format-security -nostdlibinc -fdebug-info-for-profiling -march=armv8-a -mcpu=cortex-a53 -target aarch64-linux-android10000 -DANDROID_STRICT -fPIC -Wsign-promo -Wimplicit-fallthrough -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -Wno-gnu-include-next -fvisibility-inlines-hidden  -Iframeworks/base/core/jni -Iframeworks/base/core/jni/include -Iexternal/skia/include/private -Iframeworks/base/media/jni -Isystem/media/camera/include -Isystem/media/private/camera/include -Iframeworks/base/core/jni -Iout/soong/.intermediates/frameworks/base/core/jni/libandroid_runtime/android_arm64_armv8-a_cortex-a72_shared/gen/aidl -D__LIBANDROIDICU_API__=10000 -D__LIBDL_ANDROID_API__=10000 -D__LIBC_API__=10000 -D__LIBM_API__=10000 -D__LIBDL_API__=10000 -Ibionic/libc/platform -Ipackages/modules/DnsResolver/include/dnsproxyd_protocol -Iframeworks/base/core/jni/include_vm -Ilibnativehelper/include_jni -Ibionic/libc/async_safe/include -Isystem/logging/liblog/include -Iframeworks/native/libs/binderthreadstate/include -Isystem/memory/libmeminfo/libdmabufinfo/include -Iexternal/giflib -Iframeworks/native/libs/binder/include -Isystem/libbase/include -Iexternal/fmtlib/include -Iframeworks/native/libs/binder/ndk/include_cpp -Isystem/core/libcutils/include -Isystem/core/libutils/include -Isystem/unwinding/libbacktrace/include -Isystem/core/libprocessgroup/include -Isystem/core/libsystem/include -Iout/soong/.intermediates/frameworks/native/libs/binder/libbinder/android_arm64_armv8-a_cortex-a72_shared/gen/aidl -Iout/soong/.intermediates/frameworks/native/libs/gui/libgui_window_info_static/android_arm64_armv8-a_cortex-a72_static/gen/aidl -Ibionic/libc/seccomp/include -Isystem/core/libgrallocusage/include -Iexternal/scrypt/lib/crypto -Ipackages/modules/StatsD/lib/libstatssocket/include -Iexternal/skia/android -Iexternal/skia -Iexternal/skia/client_utils/android -Iexternal/skia/include/android -Iexternal/skia/include/c -Iexternal/skia/include/codec -Iexternal/skia/include/config -Iexternal/skia/include/core -Iexternal/skia/include/docs -Iexternal/skia/include/effects -Iexternal/skia/include/encode -Iexternal/skia/include/gpu -Iexternal/skia/include/pathops -Iexternal/skia/include/ports -Iexternal/skia/include/svg -Iexternal/skia/include/utils -Iexternal/skia/include/utils/mac -Iexternal/skia/modules/particles/include -Iexternal/skia/modules/skottie/include -Iexternal/skia/modules/skparagraph/include -Iexternal/skia/modules/skshaper/include -Iexternal/skia/modules/svg/include -Iexternal/vulkan-headers/include -Ilibnativehelper/header_only_include -Ilibnativehelper/include -Ilibnativehelper/include_platform -Ilibnativehelper/include_platform_header_only -Isystem/libziparchive/incfs_support/include -Isystem/libziparchive/include -Iexternal/googletest/googletest/include -Iexternal/libcxxabi/include -Iexternal/harfbuzz_ng/src -Iframeworks/minikin/include -Iexternal/icu/libandroidicu/include -Iexternal/zlib -Iout/soong/.intermediates/frameworks/base/media/android.media.audio.common.types-V1-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/native/libs/permission/framework-permission-aidl-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/av/media/libaudioclient/audioclient-types-aidl-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/av/av-types-aidl-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/av/media/libshmem/shared-file-region-aidl-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/av/media/libaudioclient/effect-aidl-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/av/media/libaudioclient/audioflinger-aidl-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/av/media/libaudioclient/audiopolicy-types-aidl-cpp-source/gen/include -Iout/soong/.intermediates/frameworks/av/media/libaudioclient/spatializer-aidl-cpp-source/gen/include -Isystem/libhidl/base/include -Isystem/libhidl/transport/include -Isystem/libfmq/base -Isystem/libhwbinder/include -Iout/soong/.intermediates/system/libhidl/transport/manager/1.0/android.hidl.manager@1.0_genc++_headers/gen -Iout/soong/.intermediates/system/libhidl/transport/manager/1.1/android.hidl.manager@1.1_genc++_headers/gen -Iout/soong/.intermediates/system/libhidl/transport/manager/1.2/android.hidl.manager@1.2_genc++_headers/gen -Iout/soong/.intermediates/system/libhidl/transport/base/1.0/android.hidl.base@1.0_genc++_headers/gen -Iout/soong/.intermediates/hardware/interfaces/camera/common/1.0/android.hardware.camera.common@1.0_genc++_headers/gen -Iout/soong/.intermediates/hardware/interfaces/graphics/common/1.0/android.hardware.graphics.common@1.0_genc++_headers/gen -Iout/soong/.intermediates/hardware/interfaces/camera/device/3.2/android.hardware.camera.device@3.2_genc++_headers/gen -Iframeworks/native/libs/battery -Iframeworks/libs/net/common/netd/libnetdutils/include -Isystem/memory/libmemtrack/include -Iframeworks/base/libs/androidfw/include -Isystem/incremental_delivery/incfs/util/include -Isystem/core/libappfuse/include -Iexternal/boringssl/src/include -Isystem/core/debuggerd/include -Isystem/core/debuggerd/common/include -Iframeworks/native/libs/binder/ndk/include_ndk -Iframeworks/native/libs/binder/ndk/include_platform -Iframeworks/native/libs/ui/include -Iframeworks/native/libs/ui/include_private -Iframeworks/native/libs/nativebase/include -Iframeworks/native/libs/nativewindow/include -Ihardware/libhardware/include -Isystem/media/audio/include -Ipackages/modules/Bluetooth/system/types -Iframeworks/native/libs/math/include -Iframeworks/native/libs/ui/include_mock -Iframeworks/native/libs/ui/include_types -Iframeworks/native/libs/arect/include -Iout/soong/.intermediates/hardware/interfaces/common/aidl/android.hardware.common-V2-ndk-source/gen/include -Iout/soong/.intermediates/hardware/interfaces/graphics/common/aidl/android.hardware.graphics.common-V3-ndk-source/gen/include -Iout/soong/.intermediates/hardware/interfaces/graphics/common/1.1/android.hardware.graphics.common@1.1_genc++_headers/gen -Iout/soong/.intermediates/hardware/interfaces/graphics/common/1.2/android.hardware.graphics.common@1.2_genc++_headers/gen -Iout/soong/.intermediates/hardware/interfaces/graphics/mapper/4.0/android.hardware.graphics.mapper@4.0_genc++_headers/gen -Iframeworks/native/libs/gralloc/types/include -Iframeworks/native/libs/graphicsenv/include -Iframeworks/native/libs/gui/include -Isystem/libhidl/transport/token/1.0/utils/include -Iout/soong/.intermediates/hardware/interfaces/media/1.0/android.hardware.media@1.0_genc++_headers/gen -Iout/soong/.intermediates/hardware/interfaces/graphics/bufferqueue/1.0/android.hardware.graphics.bufferqueue@1.0_genc++_headers/gen -Iout/soong/.intermediates/hardware/interfaces/graphics/bufferqueue/2.0/android.hardware.graphics.bufferqueue@2.0_genc++_headers/gen -Iout/soong/.intermediates/frameworks/native/libs/gui/libgui_aidl_static/android_arm64_armv8-a_cortex-a72_static/gen/aidl -Iout/soong/.intermediates/frameworks/native/libs/gui/sysprop/ [... cut ...]
frameworks/base/core/jni/android_hardware_Camera.cpp:599:71: error: too many arguments to function call, expected 2, have 3
    status_t rc = Camera::getCameraInfo(cameraId, overrideToPortrait, &cameraInfo);
                  ~~~~~~~~~~~~~~~~~~~~~                               ^~~~~~~~~~~
frameworks/av/camera/include/camera/CameraBase.h:129:26: note: 'getCameraInfo' declared here
    static status_t      getCameraInfo(int cameraId,
                         ^
3 errors generated.
08:30:35 ninja failed with: exit status 1

#### failed to build some targets (02:33 (mm:ss)) ####
](url)

Assunzain, 2023-04-12:
try to commit with this https://github.com/Assunzain/frameworks_base/commit/c1636a8d2f29c8b9cea7f553166443dd49dfd71d

mihaimacarie98, 2023-04-12:
> try to commit with this [Assunzain/frameworks_base@c1636a8](https://github.com/Assunzain/frameworks_base/commit/c1636a8d2f29c8b9cea7f553166443dd49dfd71d)

How/where should I put this?

mihaimacarie98, 2023-04-12:
> try to commit with this [Assunzain/frameworks_base@c1636a8](https://github.com/Assunzain/frameworks_base/commit/c1636a8d2f29c8b9cea7f553166443dd49dfd71d)

I've tried but I get other errors now:
`frameworks/base/core/jni/android_hardware_Camera.cpp:482:34: error: use of undeclared identifier 'CAMERA_CMD_LONGSHOT_ON'
        rc = camera->sendCommand(CAMERA_CMD_LONGSHOT_ON, 0, 0);
                                 ^
frameworks/base/core/jni/android_hardware_Camera.cpp:484:34: error: use of undeclared identifier 'CAMERA_CMD_LONGSHOT_OFF'
        rc = camera->sendCommand(CAMERA_CMD_LONGSHOT_OFF, 0, 0);
                                 ^
frameworks/base/core/jni/android_hardware_Camera.cpp:500:29: error: use of undeclared identifier 'CAMERA_CMD_HISTOGRAM_SEND_DATA'
   rc = camera->sendCommand(CAMERA_CMD_HISTOGRAM_SEND_DATA, 0, 0);
                            ^
frameworks/base/core/jni/android_hardware_Camera.cpp:515:32: error: use of undeclared identifier 'CAMERA_CMD_HISTOGRAM_ON'
      rc = camera->sendCommand(CAMERA_CMD_HISTOGRAM_ON, 0, 0);
                               ^
frameworks/base/core/jni/android_hardware_Camera.cpp:517:32: error: use of undeclared identifier 'CAMERA_CMD_HISTOGRAM_OFF'
      rc = camera->sendCommand(CAMERA_CMD_HISTOGRAM_OFF, 0, 0);
                               ^
frameworks/base/core/jni/android_hardware_Camera.cpp:591:71: error: too many arguments to function call, expected 2, have 3
    status_t rc = Camera::getCameraInfo(cameraId, overrideToPortrait, &cameraInfo);
                  ~~~~~~~~~~~~~~~~~~~~~                               ^~~~~~~~~~~
frameworks/av/camera/include/camera/CameraBase.h:129:26: note: 'getCameraInfo' declared here
    static status_t      getCameraInfo(int cameraId,
                         ^
frameworks/base/core/jni/android_hardware_Camera.cpp:621:47: error: too many arguments to function call, expected 5, have 6
                            targetSdkVersion, overrideToPortrait);
                                              ^~~~~~~~~~~~~~~~~~
frameworks/av/camera/include/camera/Camera.h:82:25: note: 'connect' declared here
    static  sp<Camera>  connect(int cameraId,
                        ^
frameworks/base/core/jni/android_hardware_Camera.cpp:649:71: error: too many arguments to function call, expected 2, have 3
    status_t rc = Camera::getCameraInfo(cameraId, overrideToPortrait, &cameraInfo);
                  ~~~~~~~~~~~~~~~~~~~~~                               ^~~~~~~~~~~
frameworks/av/camera/include/camera/CameraBase.h:129:26: note: 'getCameraInfo' declared here
    static status_t      getCameraInfo(int cameraId,
                         ^
frameworks/base/core/jni/android_hardware_Camera.cpp:845:34: error: use of undeclared identifier 'CAMERA_CMD_METADATA_ON'
        rc = camera->sendCommand(CAMERA_CMD_METADATA_ON, 0, 0);
                                 ^
frameworks/base/core/jni/android_hardware_Camera.cpp:847:34: error: use of undeclared identifier 'CAMERA_CMD_METADATA_OFF'
        rc = camera->sendCommand(CAMERA_CMD_METADATA_OFF, 0, 0);
                                 ^
10 errors generated.
[ 17% 6034/33626] //frameworks/base/services/incremental:service.incremental clang-tidy IncrementalService.cpp
out/soong/.intermediates/frameworks/base/services/incremental/service.incremental.proto/android_arm64_armv8-a_cortex-a72_static/gen/proto/frameworks/base/services/incremental/Metadata.pb.h:503:17: warning: operator=() does not handle self-assignment properly [bugprone-unhandled-self-assignment]
  inline Mount& operator=(const Mount& from) {
                ^
17:02:12 ninja failed with: exit status 1
`

jonas-tl, 2023-04-15:
I face the same error, then I change to use this branch and issue is gone: android-13.0.0_r30

repo init -u https://android.googlesource.com/platform/manifest -b android-13.0.0_r30

losgobbi, 2024-06-06:
Hi, all. This is very old and recently I've also build for rpi4 using Ubuntu and faced no errors.
Please, close if this is not a issue anymore.
Tks and regards.