android-rpi/device_arpi_rpi4 #118: Ethernet tethering in Android rpi 11

gopaladhith, 2023-07-17:
Hello Team,
    I am new to aosp and wanted to know whether ethernet tethering option can be enabled/how to enable ethernet tethering option in android rpi aosp build. Because i don't see the ethernet tethering option under settings -> hotspot & tethering menu.

losgobbi, 2024-04-18:
Hi, in which Android version are you trying to accomplish this?
Tks and regards.

shivam07maurya, 2026-08-17:
Problem:
When routing SIM (cellular) data through Ethernet Tethering from an Android phone to a PC, GTA Online and FiveM fail to connect to game servers — connection to server times out / fails to establish, even though general internet browsing works fine.
Working configurations (no issue):
Phone connected to a second phone's WiFi hotspot → Ethernet Tethering to PC → GTA/FiveM works
SIM data via USB Tethering → GTA/FiveM works
SIM data via phone's own WiFi Hotspot (to PC's WiFi) → GTA/FiveM works
Failing configuration:
SIM (cellular) data → Ethernet Tethering → GTA/FiveM fails to connect to server (browsing/TCP traffic works fine, only game UDP traffic seems affected)