android-rpi/device_arpi_rpi4 #123: AOSP Android Rapberry pi 4 Build error

wilsavery1, 2024-04-04:
Good day,

I trying to make a build, following the instruction post on the site.

but I keet getting the followin error.

I would like to understand what i'm doing wrong.

error.log:
**********************************************
error: ninja: 'home/sysmgr/aosp_rpi4/target/product/rpi4/obj_arm/MESON_LIBCAMERA/install/usr/local/lib/libcamera-hal.so', needed by '/home/sysmgr/aosp_rpi4/target/product/rpi4/obj_arm/SHARED_LIBRARIES/camera.libcamera_intermediates/LINKED/camera.libcamera.so', missing and no known rule to make it
**********************************************

Thank you.

wilsavery1, 2024-04-04:
sysmgr@aosp:~$ make bootimage systemimage vendorimage -j$(nproc)

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=14
PRODUCT_INCLUDE_TAGS=com.android.mainline
TARGET_PRODUCT=aosp_rpi4
TARGET_BUILD_VARIANT=userdebug
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a72
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a72
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.4.0-174-generic-x86_64-Ubuntu-20.04.6-LTS
HOST_CROSS_OS=windows
BUILD_ID=UQ1A.240205.002
OUT_DIR=/home/sysmgr/aosp_rpi4
============================================
FAILED: ninja: 'home/sysmgr/aosp_rpi4/target/product/rpi4/obj_arm/MESON_LIBCAMERA/install/usr/local/lib/libcamera-hal.so', needed by '/home/sysmgr/aosp_rpi4/target/product/rpi4/obj_arm/SHARED_LIBRARIES/camera.libcamera_intermediates/LINKED/camera.libcamera.so', missing and no known rule to make it
23:00:51 ninja failed with: exit status 1

#### failed to build some targets (7 seconds) ####

losgobbi, 2024-04-09:
Hi, the instructions file point to use the product rpi4-eng, at least for the standard git clone/repo sync.
Where did you get this product aosp_rpi4?

`// this target doesnt exist...`
`$ lunch aosp_rpi4`
`In file included from build/make/core/config.mk:359:`
`In file included from build/make/core/envsetup.mk:357:`
`build/make/core/product_config.mk:226: error: Can not locate config makefile for product "aosp_rpi4"`

Suggest to use according to the docs.
Tks!!!