android-rpi/device_arpi_rpi4 #127: How to add i2c device node？

hongbii, 2024-07-15:
Hi,
I'm trying to integrate the Android-RPi arpi-11 with the NXP PN7150 NFC module, which uses I2C from onboard GPIO pins 2 and 3, an interrupt pin using onboard GPIO 23, and an enable pin using onboard GPIO 24. This is the I2C device tree definition I am trying to add to the bcm2711-rpi-4-b.dtb, but this implementation didn't work. Can anyone help? Thanks!

```
&i2c1 {
   pinctrl-names = "default";
   pinctrl-0 = <&i2c1_pins>;
   clock-frequency = <100000>;
   status = "okay";
   pn547: pn547@28 {
      compatible = "nxp,pn547";
      reg = <0x28>;
      clock-frequency = <400000>;
      interrupt-gpios = <&gpio 23 0>;
      enable-gpios = <&gpio 24 0>;
   };
};

```

losgobbi, 2024-08-28:
Hi!! Is the NXP driver at the rpi-kernel? Looking briefly, I can see a repo from NXP with a nxp-pn5xx module, which should match the definition from this "compatible" string. You have to double check that. If it was enabled, you can check the kernel logs and look for driver logs if there is any.
Regards!

nguyenanhgiau, 2026-11-10:
> Hi, I'm trying to integrate the Android-RPi arpi-11 with the NXP PN7150 NFC module, which uses I2C from onboard GPIO pins 2 and 3, an interrupt pin using onboard GPIO 23, and an enable pin using [...]

From my perspective, the good way is to add a device tree overlay for this module, e.g., arch/arm/boot/dts/overlay/pn532-overlay.dts
```
/dts-v1/;
/plugin/;

#include <dt-bindings/interrupt-controller/irq.h>
#include "i2c-buses.dtsi"

/ {
	compatible = "brcm,bcm2711";

	fragment@0 {
		target = <&i2cbus>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			pn532: pn532@24 {
				compatible = "nxp,pn532";

				reg = <0x24>;

				interrupt-parent = <&gpio>;
				interrupts = <17 IRQ_TYPE_LEVEL_HIGH>;
				status = "okay";
			};
		};
	};

	__overrides__ {
		addr    = <&pn532>,"reg:0";
		irq_pin = <&pn532>,"interrupts:0";
	};
};
```

Then, in config.txt, you should add a line: dtoverlay=pn532,addr=0x24,irq_pin=17.

Remember to enable NFC in kernel. Below is the configuration for PN532:
```
CONFIG_NFC=y
CONFIG_NFC_NCI=y
CONFIG_NFC_HCI=y
CONFIG_NFC_PN533=y
CONFIG_NFC_PN533_I2C=y
```