android-rpi/device_arpi_rpi4 #128: build  arpi-13 and arpi-14  on rpi4 ,the bluetooth can scan and match ,but can' t connected

niangegelaile, 2024-08-12:
I compiled the arpi-13 and arpi-14 branches and both were successfully installed. But they all have the same problem, Bluetooth can match, but cannot establish a connection. Could you please help solve this problem

losgobbi, 2024-08-28:
Hi!! What is the BT device that you are trying to pair/connect? Is this occurring with other BT devices too? Suggest to capture the logcat buffers around that time, maybe using some BT tags, so we can understand better the issue (past that here too, with proper format so it could be readable). Last month I've compiled arpi, but I didn't use the BT connection.
Regards!

niangegelaile, 2024-12-09:
Hello, thank you for your reply. I have found the error log below.

03-03 10:11:12.823  1832  1832 E droid.bluetooth: [0303/101112.823104:ERROR:message_loop_thread.cc(103)] DoInThreadDelayed: message loop is null for thread bt_main_thread(-1), from pc:0x7c3e9c6868
03-03 10:11:12.823  1832  1832 E droid.bluetooth: [0303/101112.823333:ERROR:btu_task.cc(94)] do_in_main_thread: failed from pc:0x7c3e9c6868
03-03 10:11:12.823  1832  1832 E droid.bluetooth: [0303/101112.823465:ERROR:bta_sys_main.cc(171)] bta_sys_sendmsg: do_in_main_thread failed

03-03 10:11:10.343  1832  1832 E droid.bluetooth: [0303/101110.343079:ERROR:hal_version_manager.cc(135)] HalVersionManager No supported HAL version
03-03 10:11:10.589  1832  1832 W ActivityThread: Failed to find provider info for bluetooth_legacy.provider (user not unlocked)

losgobbi, 2024-12-13:
Tks for the logs, but they are not very helpful due their meaning (at least for me). Anyway, I was able to test this locally with two different devices....a linux pc and a external headset. Both are not working, so BT state looks broken to me. The behavior is this:

- the scan is working, I can see my pc and the headset;
- when the connection starts, it looks like it was able to do part of the "bt handshake" (when used the pc, I can also see the pin code);
- but in the next stages of the "bt handshake", for some reason, the connection is aborted (?);
- and after that, the BT device stays as a "save device" inside settings app;

I was checking the logs like this: `logcat -b all | grep -iE "btm|l2|bluetooth"`, but I don't know yet in which layer the problem is.
I'll further debug this when I've some time.
Tks.

losgobbi, 2025-01-16:
@niangegelaile, I was debugging a little longer here but without much progress. Recently, I was trying to compare the behavior of this repo vs the https://github.com/raspberry-vanilla/android_device_brcm_rpi4. At vanilla case, BT is working from my tests. So, I'm not completely sure about where the problem is: firmware scope, kernel image/dtb`s, "product device configuration" (device folder at aosp tree) or at this particular android tag.

I was trying to pursuit that question mixing the device_arpi_rpi4 repo inside the vanilla tree, looking for better results, but I've faced a loop of different problems. Since I don't known all the history of changes of the android-rpi 14 repo, it becomes even more complex.

My suggestion is: you can try to compare yourself and debug further, maybe continue to compare with vanilla project.
a sidenote: despite the BT problem, I've faced a variation of https://github.com/android-rpi/device_arpi_rpi5/issues/1 for wifi tests. At my tests, Wifi at Android 14 doesn't connect, it points to a password error. Maybe this strange behavior is related to the BT issue too, since I think, the IC at rpi is the same, despite the software stack.

Maybe, @peyo-hd , can help, since his the owner of the repo and knows all the history of changes.
Tks and regards.